Pi4J/pi4j-v1 #121: Random fake readings on digital inputs

mosig, 2015-01-22:
I tried the Listener example founded at this link: http://pi4j.com/example/listener.html that monitors the state change of a digital input. 
I implements listening to other 3 inputs for a total of 4 digital inputs.

I follow all install instruction on Pi4j site, in fact, when I launch the application everything works correctly, and I can see the println on the terminal of the actual state change of the inputs.
I force the input state by button switch, wired with the pull down schema showed on the Pi4j site. 

After a random time, without any button pressed, the program shows me the change status on the inputs randomly, that no one has caused...

Initially I thought it was an electrical problem that my buttons created "noise" but I checked out the wire connection and the buttons work properly.
These false readings happen sporadically but can not seem to stop for as long as the program is running.

I'm running the 2014-09-09 version of Raspbian.
Any thoughts on this?

savageautomate, 2015-02-12:
It really sounds like the pin is floating and not biased with either the internal PULL-UP or PULL-DOWN resistance or externally with physical PULL-UP or PUL-DOWN resistors in your circuit. 

If numerous flip flops occurred around the actual push/release of the physical buttons I would think its more of a debounce problem.

mosig, 2015-03-17:
Hi Robert,
in my source code I get my gpio controller:
<< GpioController gpio = GpioFactory.getInstance(); >>
and set up the pin with the following code:
<< final GpioPinDigitalInput lightGreen = gpio.provisionDigitalInputPin(RaspiPin.GPIO_00, PinPullResistance.PULL_DOWN); >> as is shown in the example...

As you've highlighted it seems an instability of the pin, but the above statement is not enough to set the pull down of the pin? The connection is completly clean and it is not possible having problems like debouncing or electrical interference.

Today i'm running Raspberry 2 Model B, with the 2015-02-16 version of Raspbian, but I have not noticed any improvement.

How do I proceed to identify the problem?

savageautomate, 2015-03-17:
Can you post your sample code and I will run it for an extended period to see if I can reproduce. 

Thanks, Robert

mosig, 2015-03-19:
Of course!

// create gpio controller
Thread.sleep(1000);
GpioController gpio = GpioFactory.getInstance();

final GpioPinDigitalInput lightGreen = gpio.provisionDigitalInputPin(RaspiPin.GPIO_00,               PinPullResistance.PULL_DOWN);

lightGreen.addListener(new GpioPinListenerDigital() {
           @Override
            public void handleGpioPinDigitalStateChangeEvent(GpioPinDigitalStateChangeEvent event)   {
                try {
                    update.updateGreenLight(event.getState().toString());
                } catch (Exception ex) {
                    Logger.getLogger(IFLDuePuntoZero.class.getName()).log(Level.SEVERE, null, ex);
                }
                // display pin state on console
                System.out.println(" --> Light Green: " + event.getPin() + " = " + event.getState());
            }
            
        });

Where the object "update" is calling a function that execute an http get to a web service.

nilsschmidthamburg, 2016-09-08:
Hi @mosig, 
did you manage to figure out what caused the issue?

We are having the same problem over here with the exact code taken from here: http://pi4j.com/example/listener.html on a RaspberryPi 3 running Raspbian Jezzy.

We couldn't manage to get rid of the random 'noise' thus we had to go away from the hardware based interrupt model with Listeners to a rather stupid polling mode. 

What I've figured out so far and considered quite interesting is the following when recieving a GpioPinDigitalStateChangeEvent event:

- event.getState() => changes with each event (either HIGH or LOW)
- ((GpioPinDigitalInput) event.getPin()).getState() is always LOW (which is the correct button state)

Let's hope we get this thing solved. 
Regards
Nils

mosig, 2016-09-09:
Hi @nilsschmidthamburg in the past we resolve the problem adding some electronic components on the circuit...

nilsschmidthamburg, 2016-09-09:
Thanks for the quick reply! 
I see, using external pull down resistors solves the problem for us as well. But we are aiming for a solution to avoid this, which currently leads us to the polling model... :-/

savageautomate, 2016-09-09:
@nilsschmidthamburg 

Can you determine (by measurement) if the internal resistors are being applied property when Pi4J provisions a pin?  If possible measure the pin with nothing else connected to it.   Perhaps also try changing the PULL UP / PULL DOWN states when Pi4J provisions the pin to make sure its working properly.  

The symptom sure sounds like a floating pin.  Maybe something has changed and the code is not properly setting the pull up/down state any longer or on certain hardware?

Thanks, Robert

nilsschmidthamburg, 2016-09-09:
Hi @savageautomate,
I would love to help on this, so I can defiantly try to measure if the resistors are applied. Only problem is, (and now comes the embarrassing part) I am not very good with electronics, so you might need to give me a hint what exactly (or how exactly) to meassure that. I do have a voltmeter/amperemeter and roughly know how to use it:-)

Regards
Nils

savageautomate, 2016-09-09:
First, which GPIO pin number are you attempting to use?  The shared I2C pins have hard-wired pull-up resistors.   Use this chart for GPIO numbering: http://pi4j.com/pins/model-3b-rev1.html

If not using an I2C pin, the GPIO pin resistance should be somewhere around 50K Ohms between the pin and GROUND when the pin is configured for PULL-DOWN or around 50K Ohms between the pin and +3VDC when the pin is configured for PULL-UP.

Thanks, Robert

sramilli, 2016-10-06:
Hi,
had the same problem and spent many weeks troubleshooting it. 
In the end i´m pretty sure that my board is faulty on one pin and there is no way for a work-around. The only solution is to substitute the board (that i will do in the next days).
If this can help anyone getting out the frustration i write down my experience below.

My system:
Raspberry (1) Pi B rev 2, 
GPRS modem attached to the UART,
2 channel relay with optocoupler attached to 5V and a GPIO pin of the Pi,
a custom prorotypal PCB board with 5 LEDS and 2 buttons attached to the Pi´s GPIO.
One button is used to shut down the pi gracefully via software, the other one is used to change operational mode. Both buttons have external pull-down resistors + 1k resistor as protection in imput to the pi + a small capacitor for debouncing.

The probelm:
Sometimes when i came home the pi was off, because it detected a false button press on the "OFF button" so i started watching the PCB trying to spot a bad soldering, but i didnt find any and throwed it away and soldered a new one. 
That solved the problem, but after 6 months the problem started again!! I got very frustrated i can tell. Now the Pi shutted down when the GSM module were receiving SMS´s. Crap!

Tests:
I swapped the wires on the PCB of the two buttons but i had the same problem so it wasnt a button failure nor the pull-down resistor or capacitor.
I soldered an extra wire to another spot on the PCB´s ground to another ground pin of the Pi, just to be extra sure.
I swapped the wires at both ends, but i had the same problem, so it wasnt the wires either.
I switchet on the internal pull-down as well, but the problem persisted. And this surprised me a lot!
I switched pin assignation in the software for the two buttons and then i was instead having the problem that the operational mode was switching randomly.

Notes:
When the Pi failed if failed continuously 10 times out of 10, and if i moved the components around a littlebit it maybe started to work or maybe not...
When it worked i jus carefully unplugged the wire from the PCB of the button that experienced before the problem and sended sms´s and that caused the same problems.

Conclusion:
The pin was floating even if it was attached to ground, meaning the Pi was faulty.

savageautomate, 2021-01-15:
**Closed.**  No further activity.   To my knowledge all input pins are working correctly in the latest Pi4J versions.

closed 2021-01-15 by savageautomate