Pi4J/pi4j-v1 #124: State-switching at initialisation

K1LLUM1N471, 2015-01-26:
Hello,

I'm using an addon-board with four opening relays. Everything works fine and I can switch them all correctly.
To initialise I used "relais1 = gpio.provisionDigitalOutputPin(RaspiPin.GPIO_00, "", PinState.HIGH);"
When I run my sources a second time and the relay is already is the state HIGH, the first relay change it's state to LOW for a few milliseconds -> that is not desired!
Is this issue known and does a workaround exists?

Kind regards,
K1LLUM1N471

savageautomate, 2015-02-11:
Issue confirmed and fixed.  I'll post a new build tonight with this fix.

closed 2015-02-11 by savageautomate

savageautomate, 2015-02-24:
Hi K1LLUM1N471 ,

Were you able to verify if the new build fixed this issue for you?

Thanks, Robert

K1LLUM1N471, 2015-03-26:
Thank you, everything seems to work fine now. 
Yesterday I verified the functionality with an oscilloscope, too.
Kind regards,
K1LLUM1N471

savageautomate, 2015-03-26:
Thanks for the update confirmation!