Pi4J/pi4j-v1 #125: Maven artifact cannot be included in a shaded jar

rarkenin, 2015-01-31:
When including pi4j in Maven and creating a shaded JAR using maven-shade-plugin, the shader tries to open `pi4j-native-0.0.5-hard-float.so` as if it were a JAR file, to unpack it, which is not possible since a .so file is not a jar file. The `pom.xml` of pi4j should be amended to prevent this (as shading other native libraries is possible out of the box).

savageautomate, 2015-02-11:
In your maven-shade-plugin configuration you can exclude the native project.

```
	            <configuration>
	              <shadedArtifactAttached>true</shadedArtifactAttached>
	              <artifactSet>
	                <excludes>
	                  <exclude>com.pi4j:pi4j-native</exclude>
	                </excludes>
	              </artifactSet>
	            </configuration>
```

savageautomate, 2015-02-12:
Ok, I found a better way to address this problem.  It should be fixed now and will be included in the next build.

commit 25848c02 (2015-02-12): updated fix #125; turns out we can't remove the pi4j-native dependency as it affects the project build order.  added it back and set the scope to provided.
closed 2015-02-12 by savageautomate