Pi4J/pi4j-v1 #126: Wrong GPIO Info

metropt, 2015-02-11:
Hi,
The GPIO image you have here http://pi4j.com/pins/model-b-plus.html is incorrect.

savageautomate, 2015-02-11:
Can you please elaborate?  What is wrong so that we can correct it.  
Please keep in mind that the pin numbering scheme is based on the WiringPi numbering scheme and is not the raw Broadcom chip GPIO pin numbers.

metropt, 2015-02-11:
Hmm, I don't know how wirinpi works but the GPIO from the Raspberry shecmatic is the following: http://www.element14.com/community/servlet/JiveServlet/previewBody/68203-102-6-294412/GPIO.png per example the GPIO 23 and 24 are diferent on your image compared to the original.

savageautomate, 2015-02-11:
Yep, those are the raw broadcom chip GPIO pin numbers.  The Pi4J project uses the (abstracted) WiringPi numbering scheme.  The image below shows a mapping between the two different schemes. 

[image]

Thanks!

closed 2015-02-11 by savageautomate