Pi4J/pi4j-v1 #127: GPIO not working anymore @ Raspberry Pi 2 (2015 Model)

StefanWokusch, 2015-02-11:
Hey there,

we've tried to upgrade to the newest Raspberry Pi 2 model - I guess you guys have heard ;-)!
Unfortunately it seems, that Pi4J won't work anymore with the new RPi. We've tried to use the 0.0.5 and the 1.0-SNAPSHOT releases - no luck here.

Some guys at the RaspberryPi forum wrote, that there was a change in hardware adresses for GPIO usage? (see http://www.raspberrypi.org/forums/viewtopic.php?f=33&t=98740)

- Did you experience the same issue (no GPIO connectivity with Pi4J) when using the latest model?`
- Maybe you can tell us where to found the raw GPIO hardware addresses in your native project/C classes?

Thanks for your work.
Best regards.

savageautomate, 2015-02-11:
Based on Gordon's post on WiringPi.com, it looks like Pi4J will need to be rebuilt using the latest WiringPi build.
http://wiringpi.com/wiringpi-updated-for-the-new-pi-v2/

You can build Pi4J yourself which can automatically pull in the latest WiringPi and build it but you will need to include the native part of the build.

'mvn clean install -P native -Dpi.host={your Pi's IP Address}'

Or you can wait until I post a new build, which hopefully will be this weekend.

Thanks, Robert

savageautomate, 2015-02-12:
I have not received my hardware yet, so I am not able to test the latest build on Raspberry Pi 2 model B.
However, I have published a new 1.0-SNAPSHOT build that should include the updates from WiringPi to support the new model.   If you have time, please give it a try and post back your results.  

Thanks, Robert

StefanWokusch, 2015-02-12:
Thanks a lot. Seems working fine on the new Reapberry Pi now.

closed 2015-02-12 by StefanWokusch

savageautomate, 2015-02-12:
Excellent.  Thanks for the feedback.