Pi4J/pi4j-v1 #130: Problem with SPI mode

entrusc, 2015-02-15:
When using `Spi.wiringPiSPISetupMode()` I get an UnsatisfiedLinkError:

    Exception in thread "main" java.lang.UnsatisfiedLinkError: com.pi4j.wiringpi.Spi.wiringPiSPISetupMode(III)I
            at com.pi4j.wiringpi.Spi.wiringPiSPISetupMode(Native Method)
            at de.darkblue.pi.rf24.RF24.<init>(RF24.java:133)
            at de.darkblue.pi.pitest.App.main(App.java:18)

 Obviously the underlying wiringpi lib is not up to date. I use the `1.0-SNAPSHOT` version of Pi4j from the maven repo.

entrusc, 2015-02-15:
On closer inspection the problem seems to be that in the libpi4j.so the java specific function `Java_com_pi4j_wiringpi_Spi_wiringPiSPISetupMode` is missing but the wiringpi method `wiringPiSPISetupMode` is already there:

	 $ readelf -Ws ~/libpi4j.so | grep PiSPI
		77: 00006930   200 FUNC    GLOBAL DEFAULT   12 Java_com_pi4j_wiringpi_Spi_wiringPiSPIDataRW__I_3SI
		85: 00006728   132 FUNC    GLOBAL DEFAULT   12 Java_com_pi4j_wiringpi_Spi_wiringPiSPIDataRW__ILjava_lang_String_2I
	   109: 000067ac   388 FUNC    GLOBAL DEFAULT   12 Java_com_pi4j_wiringpi_Spi_wiringPiSPIDataRW__I_3BI
	   125: 000069f8    12 FUNC    GLOBAL DEFAULT   12 Java_com_pi4j_wiringpi_Spi_wiringPiSPISetup
	   189: 00006720     8 FUNC    GLOBAL DEFAULT   12 Java_com_pi4j_wiringpi_Spi_wiringPiSPIGetFd
	   226: 0000a4d0   388 FUNC    GLOBAL DEFAULT   12 wiringPiSPISetupMode
	   250: 0000a654     8 FUNC    GLOBAL DEFAULT   12 wiringPiSPISetup
	   260: 0000a468   104 FUNC    GLOBAL DEFAULT   12 wiringPiSPIDataRW
	   310: 0000a450    24 FUNC    GLOBAL DEFAULT   12 wiringPiSPIGetFd
	   270: 00000000     0 FILE    LOCAL  DEFAULT  ABS wiringPiSPI.c
	   432: 00006930   200 FUNC    GLOBAL DEFAULT   12 Java_com_pi4j_wiringpi_Spi_wiringPiSPIDataRW__I_3SI
	   452: 000067ac   388 FUNC    GLOBAL DEFAULT   12 Java_com_pi4j_wiringpi_Spi_wiringPiSPIDataRW__I_3BI
	   578: 0000a468   104 FUNC    GLOBAL DEFAULT   12 wiringPiSPIDataRW
	   602: 0000a654     8 FUNC    GLOBAL DEFAULT   12 wiringPiSPISetup
	   603: 00006728   132 FUNC    GLOBAL DEFAULT   12 Java_com_pi4j_wiringpi_Spi_wiringPiSPIDataRW__ILjava_lang_String_2I
	   615: 000069f8    12 FUNC    GLOBAL DEFAULT   12 Java_com_pi4j_wiringpi_Spi_wiringPiSPISetup
	   626: 0000a450    24 FUNC    GLOBAL DEFAULT   12 wiringPiSPIGetFd
	   640: 0000a4d0   388 FUNC    GLOBAL DEFAULT   12 wiringPiSPISetupMode
	   686: 00006720     8 FUNC    GLOBAL DEFAULT   12 Java_com_pi4j_wiringpi_Spi_wiringPiSPIGetFd

savageautomate, 2015-02-15:
Whoops.  Looks like I missed a step.  Fixing it now!

closed 2015-02-15 by savageautomate

savageautomate, 2015-02-15:
1.0-SNAPSHOT rebuilt & redeployed

commit b725f1c9 (2015-02-15): fixed #130; adding missing JNI method 'Java_com_pi4j_wiringpi_Spi_wiringPiSPISetupMode'
commit 1d67cc6f (2015-02-26): fixed #130; adding missing JNI method 'Java_com_pi4j_wiringpi_Spi_wiringPiSPISetupMode'