Pi4J/pi4j-v1 #131: Use of two PiFaces Digitals  does not work correclty

thorsten-koch, 2015-02-18:
Hi, 

I am using two PiFace Digitals via the PiFace Rack. However, it is not possible to address the inputs and output of each devices independently. While going through the code, I found out that the parameter spiAddress in the class PiFaceGpioProvider is never assigned to the variable address. 
I add a corresponding assignment and calculated the new address [0-4] as follows 

private byte getSPIControlByte(byte hardwareAddress) {
	byte board_addr_pattern = (byte) ((hardwareAddress << 1) & 0xE);
	return (byte) (PiFace.DEFAULT_ADDRESS | board_addr_pattern);
}

However, this didn't solve my problem. If I set the LEDs of one board, the LEDs of the other board are set as well.

Any idea how to solve this

savageautomate, 2015-02-22:
I have just published a new 1.0-SNAPSHOT build.  Can you try and see if the SPI address fix helps this issue.  

Thanks, Robert

thorsten-koch, 2015-02-26:
I downloaded the latest version from maven and got the following exception, when I create a new PiFace object:
Exception in thread "main" com.pi4j.io.gpio.exception.InvalidPinModeException: Invalid pin mode on pin [INPUT 1 (SWITCH 1)]; cannot setState() when pin mode is [input]
at com.pi4j.io.gpio.GpioProviderBase.setState(GpioProviderBase.java:173)
        at com.pi4j.gpio.extension.piface.PiFaceGpioProvider.setState(PiFaceGpioProvider.java:310)
        at com.pi4j.io.gpio.GpioProviderBase.export(GpioProviderBase.java:74)
        at com.pi4j.io.gpio.impl.GpioPinImpl.export(GpioPinImpl.java:155)
        at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionPin(GpioControllerImpl.java:514)
        at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionPin(GpioControllerImpl.java:493)
        at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionPin(GpioControllerImpl.java:488)
        at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionDigitalInputPin(GpioControllerImpl.java:594)
        at com.pi4j.device.piface.PiFaceBase.<init>(PiFaceBase.java:68)
        at com.pi4j.device.piface.impl.PiFaceDevice.<init>(PiFaceDevice.java:45)

To ensure, that it does not depend on my program, I downloaded the latest *.deb file and compiled the PiFaceExample.java. However, I got the same exception as above.

savageautomate, 2015-02-27:
I'll try this out today!  It could be some bug introduced ....

savageautomate, 2015-03-01:
Found the issue and this was a casualty of another recent fix.  Pushing a new 1.0-SNAPSHOT build out now with this fix.

thorsten-koch, 2015-03-05:
Thanks for your work. 
However, until now it is still not possible to address the PiFaces correctly. 

If you switch the jumper JP1 and JP2 to get another of the possible 4 hardware addresses, the PiFace does not react correctly. For example, outputs are set although the PiFace address and the one of the program do not match. 

I use the following code for the initialization

<code> PiFace PIFACE_BOARD_1 = new PiFaceDevice(getSPIControlByte((byte) 1), Spi.CHANNEL_0); </code>

<code> private static byte getSPIControlByte(byte hardwareAddress) { </code>
<code>	byte board_addr_pattern = (byte) ((hardwareAddress << 1) & 0xE);</code>
<code>	return (byte) (PiFace.DEFAULT_ADDRESS | board_addr_pattern);</code>
<code>}</code>

savageautomate, 2015-03-05:
thorsten-koch,

I don't have multiple PiFace devices to test simultaneously; however I did find a problem when attempting to set the PiFace chip address to something other than the default address.  I have fixed that issue and I tested my single PiFace using the four separate hardware address combinations.  Please try the latest 1.0-SNAPSHOT build -- I just deployed a new build.  

Thanks, Robert

closed 2015-03-05 by savageautomate

thorsten-koch, 2015-03-06:
Hi Robert, 

I tested it with two devices and its working now. 
Thanks, Thorsten

savageautomate, 2015-03-06:
Thorsten,

Thank you for for the confirmation update.

Cheers, 
Robert

commit 40258fb7 (2015-02-22): attempting to fix #131; spi address parameter was not getting assigned in constructor
commit 715bcbcc (2015-02-26): attempting to fix #131; spi address parameter was not getting assigned in constructor
commit c0ecc568 (2015-03-01): fixed #131; a recent fix in commit 0d021d8 caused this bug for provisioning input pins such as for PiFace impl.
commit 00aeefb5 (2015-03-05): fixed #131; added SEQOP bit to IOCON register to handle chip hardware addressing.