Pi4J/pi4j-v1 #132: pi4j Getting wierd characters on QC2004A

aphtk, 2015-02-18:
I am having the same issue as --> https://github.com/Pi4J/pi4j/issues/13

I am pretty sure the wiring is correct. 
I am using WiringPiLCDExample.java as my base.
I am setting the LCD to 8-bit mode
I have the following setup using wiringpi numbering convention (ref. to http://pi4j.com/pins/model-b-rev2.html):
D4 = 2 // gpio pin 13
D5 = 3 // gpio pin 15
D6 = 4 // gpio pin 16
D7 = 5 // gpio pin 18

I have the p5 header soldered and tested to work and using:
RS = 17 
E   = 18

I am pretty sure that LCD panel is working correctly, as it was tested with adafruit's python library (although I could never get extended ASCII to show).

When I run the code (with the required pin # changes), all I am getting is some "o" and "->" and blinking cursor at the end.

Can you let me know what I might be doing wrong here?

savageautomate, 2015-03-01:
Hi aphtk,

If you are using 8 bit LCD you need individual pins connected and assigned D0 thru D3.  

Here is a pictorial diagram from Gordon's WiringPi site for 8 bit connected LCD.
![Image of 8-bit LCD]
(https://projects.drogon.net/wp-content/uploads/2012/07/lcd8_bb.png)

More details here:
http://wiringpi.com/dev-lib/lcd-library/

closed 2015-03-01 by savageautomate

aphtk, 2015-03-02:
Thanks for your reply. Found that inspite of whats on the package qc2004a is a 4bit LCD. Also p5 header did not work. Std. pins worked.