Pi4J/pi4j-v1 #133: java.lang.ExceptionInInitializerError Introduced in in Maven Build 2/20/2015 1.0-SNAPSHOT

jorgejordan, 2015-02-20:
Error:
java.lang.ExceptionInInitializerError: null
	at com.pi4j.io.gpio.RaspiPin.createDigitalPin(RaspiPin.java:84)
	at com.pi4j.io.gpio.RaspiPin.<clinit>(RaspiPin.java:45)

Offending line of code:
private final transient static Pin EXTERNAL_POWER_PIN = RaspiPin.GPIO_29;

Maven Dependency

		<dependency>
			<groupId>com.pi4j</groupId>
			<artifactId>pi4j-core</artifactId>
			<version>1.0-SNAPSHOT</version>
			<exclusions>
				<exclusion>
					<groupId>com.pi4j</groupId>
					<artifactId>pi4j-native</artifactId>
				</exclusion>
			</exclusions>
		</dependency>

I've been building fine for a while using the 1.0-SNAPSHOT (I'm building on my LAPTOP), but tried to build this morning and got the above error.  I don't think anything changed, but noticed there was a new SNAPSHOT release.

jorgejordan, 2015-02-20:
Can work around by not creating the static definition, and just passing directly to GpioController.provisionDigitalInputPin

closed 2015-02-20 by jorgejordan

savageautomate, 2015-02-21:
This issue has been fixed and a new 1.0-SNAPSHOT has been published.

commit 7b557d96 (2015-02-21): fixed #133; pins collections was NULL throwing null pointer exception.
commit 801c7e8d (2015-02-26): fixed #133; pins collections was NULL throwing null pointer exception.