Pi4J/pi4j-v1 #135: Serial Communication byte errors

tj128, 2015-02-21:
Hi there, Great work on the library, it works great and is easy to use, however I am stuck on one problem which I believe is within the library but please correct me if I'm wrong... 

I am using a raspberry pi with the pi4j Serial communication library to try and communicate with a separate pc. I am sending and receiving data in both directions, but when looking at the data in hex representation there are constant errors that occur.

Nearly all other strings, varying in size and format work but when sending the number '8' i receive a '5' on the other end. Example sending bytes from the pi4j library using 'serial.write((byte)82); ' displays on my computer as 52.  
This is supposed to be a message going to a RFID reader which should send 'a0 03 82 00 db'. All is correct apart from the 82 which is replaced as 52??? (a0 03 52 00 db)   

Also when reading data from the serial port using the 'event.getdata()' again many examples work fine, but if an 8 is present or a letter starts the string some bytes are missing and other added. e.g. sending 'e4 04 82 00 04 92' comes out on the pi screen as 'c3 a4 04 c2 82 00 04 c2 92'. most of the data is there but is split up and random characters added!
 
I am using the Hex.encodeHexString from apache to convert from string to hex, which as far as i can tell through other testing does not have a problem with it...

Any suggestions on how to get around this problem?    I just want to view hex coming in from the serial port!

Thanks

savageautomate, 2015-02-22:
Hi tj128,

I have been working on a new serial implementation to properly support a byte level interface for working with the data.  I suggest that you try the experimental build 1.1-SNAPSHOT.  The serial API is completely overhauled and should provide far better results when working with bytes. 

GITHUB BRANCH:
https://github.com/Pi4J/pi4j/tree/serial_enhancements

DOWNLOAD:
http://get.pi4j.com/download/pi4j-1.1-SNAPSHOT.zip

Please let me know if you find any problems ... remember it is very new and experimental :-)

Sample code below:

```
import com.pi4j.io.serial.Baud;
import com.pi4j.io.serial.DataBits;
import com.pi4j.io.serial.FlowControl;
import com.pi4j.io.serial.Parity;
import com.pi4j.io.serial.Serial;
import com.pi4j.io.serial.SerialDataEvent;
import com.pi4j.io.serial.SerialDataEventListener;
import com.pi4j.io.serial.SerialFactory;
import com.pi4j.io.serial.StopBits;
import java.io.IOException;
import java.util.Date;

public class SerialExample1 {
    
    /**
     * @param args the command line arguments
     * @throws java.lang.InterruptedException
     * @throws java.io.IOException
     */
    public static void main(String[] args) throws InterruptedException, IOException {
 
        // !! ATTENTION !!
        // By default, the on-board serial port is configured as a console port 
        // for interacting with the Linux OS shell.  If you want to use 
        // the serial port in a software program, you must disable the 
        // OS from using this port.  Or you can use a USB to serial adapter.
                
        System.out.println("<--Pi4J--> Serial Communication Example ... started.");
        System.out.println(" ... connect using settings: 38400, N, 8, 1.");
        System.out.println(" ... data received on serial port should be displayed below.");

        // create an instance of the serial communications class
        final Serial serial = SerialFactory.createInstance();

        // create and register the serial data listener
        serial.addListener((SerialDataEventListener) (SerialDataEvent event) -> {
            try {
                System.out.println("[BYTES RECEIVED (" + event.length() + ")]  " +
                        event.getHexByteString());
            } catch (IOException ex) {
                System.err.println(ex.getMessage());
            }
        });
        
        try {
            // open the default serial port provided on the GPIO header
            serial.open(Serial.FIRST_USB_COM_PORT, Baud._38400, DataBits._8, 
                        Parity.NONE, StopBits._1, FlowControl.HARDWARE);

            // if your program is not going to read the received data from
            // the serial receive buffer, then disable buffering!
            //serial.setBufferingDataReceived(false);

            // continuous loop to keep the program running until the user terminates
            for (;;) {
                // write a formatted string to the serial transmit buffer
                serial.writeln("CURRENT TIME: " + new Date().toString());
                
                // wait 1 second before continuing
                Thread.sleep(1000);
            }
        }
        catch(IllegalStateException|IOException ex) {
            System.err.println(ex.getMessage());
        }
    }
}
```

tj128, 2015-02-26:
That works perfectly! Thanks, Just one more question, not sure if its another bug or my coding...  
But I'm building a gui in which to communicate with the RFID reader, and Im having having to create multiple instances of 'serial'. providing i use serial.close(); and serial.removeListener(); this should make them redundant in the program? 

the problem is I have added close and remove listener to both serial1 and serial2 which i have created, but if i run one a second time, the data is collected and displayed twice...   would there be a more suitable way in which to have multiple instances running within a program??

Thanks Again!

savageautomate, 2015-02-27:
Are you creating two instances because you are communicating via two separate serial ports?  Are the two instances running concurrently?

tj128, 2015-02-27:
Hi there, I am creating a java swing application window, which means I have to split the code up into many 'actionLIstener' events.  If i set up a serial instance at the beginning of a program and a listener, once sending serial data within an "actionListener' the Override does not function for the serial listener.   the way in which I have it to work is creating separate serial instances for each 'action performed'. this allows me to send different data for different events and set up the listener to filter the results as applicable for the specific command.    At the end of each 'action' I am using serial.removeLIstener(); and serial.close(); which to my knowledge should terminate all serial process'? However If i start the process again within the same runtime, it's adding an extra listener each time the action is preformed. this continues until my app crashes....
So I only have one serial port, but the way in which the java swing is programmed I need to set up multiple listeners for each window to allow the dataListener to be active...  is there a way in which to terminate the whole process without exiting the program?
RegardsTom

Date: Fri, 27 Feb 2015 06:32:27 -0800

savageautomate, 2015-03-01:
Multiple instance of serial when trying to reference the same port will probably fail.  The implementation for each serial instance contains a monitoring thread that is polling and picking up (reading) the serial data from the JNI layer.   So you may have two instances competing for the same data -- who knows which one will win and get the data.  

I think you should probably define the serial impl in one place and create your own mechanism for sharing the received data or sending data for multiple listener events.    

Perhaps if you provide some basic example code here of what you are trying to accomplish one of us can provide a better suggestion.

Thanks, Robert

savageautomate, 2015-03-01:
On the other hand ...
If you are performing very encapsulated/atomic communications with the serial port for each listener event where you could open the port, send your data, then close the port, then multiple serial instances may be ok.  There is a serial.shutdown() method that you could use when done with a port.  This will close the port and shutdown the monitoring thread.  

Thanks, Robert

mik0205, 2016-05-25:
Hi Robert,

thank you for this great library.

I want to use the pi4j Serial communication library to communicate with an external device with

the following communication parameters: 9600, **7**, N, 1.

I have implemented a test program and found out that the received characters are not the

same as the send characters. But if I use a terminal program with 9600, **8**, N, 1  it works!

After some investigations I found in the file com_pi4j_jni_Serial.c the following lines (from line 226)

    ...

    options.c_cflag |= CS8 ;

    // set data bits configuration
    switch(dataBits)
    {
        case DATA_BITS_5:
            options.c_cflag |= CS5 ; // 5 data bits
            break;
        case DATA_BITS_6:
            options.c_cflag |= CS6 ; // 6 data bits
            break;
        case DATA_BITS_7:
            options.c_cflag |= CS7 ; // 7 data bits
            break;
        case DATA_BITS_8:
            options.c_cflag |= CS8 ; // 8 data bits
            break;

        default:{
            throwIOException( env, "Invalid/unsupported DATA BITS configuration property");
            return -4 ;  // INVALID/UNSUPPORTED DATA BITS CONFIG
        }
    }

Is it possible that this is not correct, because the value of  options.c_cflag is initialized with  CS8

and than bitwise or with CS7. So both bits should be active after this switch case statement, isn't it?

Many thanks for your response.

savageautomate, 2016-05-25:
That certainly looks suspect.  I'll fix that and push out a new SNAPSHOT build today and we can see if that corrects the issue.  I'll post back when the new build is ready.  Thanks for finding this!  

There is also some discussion on the 7-bit issue here. 
https://groups.google.com/forum/#!topic/pi4j/N6NUh_XFabQ

But your find may be the real underlying problem.

Thanks, Robert

savageautomate, 2016-05-25:
New 1.1-SNAPSHOT build is now available to test this fix.  Thanks, Robert

savageautomate, 2016-05-25:
Related Bug: https://github.com/Pi4J/pi4j/issues/195

mik0205, 2016-06-01:
Hi Robert,

I've checked the new version and now it runs with 9600, 7, N, 1.

But when I use odd or even **parity** it does not work when I send

some data. Sending one byte is ok, but if I send 6 bytes the last

4 bytes are wrong.

I send

ABCDEF

and receive

A
B
(
T
0x94
0xfa

savageautomate, 2016-06-13:
Hi @mik0205 ,

I was able to find some time to revisit this issue and test this today using the latest 1.1-SNAPSHOT build on a Raspberry Pi 2B and it seems to be working properly for me.

I tried the following serial configurations:

- 38400, 8N1
- 38400, 8O1
- 38400, 8E1
- 38400, 7N1
- 38400, 7O1
- 38400, 7E1
- 38400, 7E2
- 9600, 7N1
- 9600, 7O1

Data packets containing multiple bytes were both transmitted and received in tact between the Raspberry Pi and another computer.

Please let me know if you are still seeing this problem and what environment/settings your are using so I can try to duplicate.

Thanks, Robert

savageautomate, 2016-06-13:
@mik0205 

So in case you were using a Raspberry Pi 3B and it's default serial port "/dev/ttyS0" this is a software-based UART (_miniUART_) and it unfortunately does not support PARITY.  

The hardware-based UART ("/dev/ttyAMA0") is assigned to the Bluetooth chip by default.  You can disable the Bluetooth functions on the Raspberry Pi 3B and use a device-tree overlay to re-map the hardware-based UART back to the GPIO header pins as it was in previous RPi models.

See the instructions on this page for details on how to configure the device-tree overlay and disable the bluetooth modem/service.  https://openenergymonitor.org/emon/node/12311 

I did test the serial port on a RPi 3B using this device-tree overlay and communicating via "/dev/ttyAMA0" port and everything seemed to be working properly.

Thanks, Robert

mik0205, 2016-06-14:
Hi Robert,

yes, I use a Raspberry Pi 3. Many thanks for your informations. I will give it a try and come back when I have results.

Thanks,  Michael

savageautomate, 2016-06-14:
If you do use serial device "/dev/ttyS0" in a project, make sure your RPi 3B has the latest updates and firmware installed.  They have made some improvements to make the miniUART work a bit better.  But still no parity support.

savageautomate, 2016-06-14:
I'll go ahead and close this issue for now.  Feel free to reopen it if you encounter any problems.
Thanks, Robert

closed 2016-06-14 by savageautomate

mik0205, 2016-06-14:
Hi Robert,
I have tested it with 9600, 7E1 and it worked great with my modified RPi 3! 
Thank you very much for your support!
And many thanks for this great library!

Michael

savageautomate, 2016-06-14:
You are welcome!  Thanks for testing this so quickly.  I'm trying to wrap up a few open issues and finally release v1.1.

commit 082772f2 (2016-05-25): attempt to fix #135; serial communication using 7 data bits
commit fdf30d7a (2016-05-25): attempt to fix #135; serial communication using 7 data bits