Pi4J/pi4j-v1 #138: I2C bus instance is destroyed in the shutdown method from GpioProvider implementations classes.

istarciuc, 2015-03-06:
Hi,
First of all I want to congratulate you with your great work!

I'm trying to write an application using PCA9685, and MCP23017 that should work dynamic (plug and play) and I had some problems to implement it because the I2C bus instance is destroyed in shutdown method from GpioProvider implementation classes. 
From my point of view, in the implementation of the GpioProvider classes, there are two aspects that i'm not agree with them.
1. The I2C bus instance should be injected using constructor like in  PCA9685GpioProvider class and should not create the I2C bus instance in the constructor like in MCP23017GpioProvider class even if it is the same instance.   
This classes should be as independent as possible from other classes.
2. The I2C bus instance should not be destroyed in the shutdown methods from GpioProvider implementations classes, this classes should not be responsible for that. 
Maybe, in some conditions, I want to destroy the MCP23017GpioProvider instace but in the same time I use I2C bus instance for PCA9685GpioProvider instance to control stepper motors.
 
This is my opinion, and to fulfill the application requirements (plug and play) I've modified the source code from this classes. 

Best regards.

savageautomate, 2015-03-06:
Sounds like a good change.  I'll take a look this evening

savageautomate, 2015-03-08:
Fixed.  I added new constructors that accept I2C bus.  If you provide the I2C bus instance in the constructor then the internal implementation won't call a bus.close() on a shutdown.

closed 2015-03-08 by savageautomate

istarciuc, 2015-03-09:
Good job. Thanks!

commit 71845b3f (2015-03-08): fixed #138; added I2C bus constructors and bus owner flag to control/prevent bus closing.