Pi4J/pi4j-v1 #142: Addition of JavaDoc comments

nkolban, 2015-03-23:
Would it be possible to add more JavaDoc comments?  For example, the utils.String package doesn't contain any descriptions of the methods.

AGausmann, 2015-09-20:
+1 for this. The documentation is severely lacking; even after browsing through the source code I haven't found the range for analog inputs. I want to assume it is a voltage level, but I can't. Units and values like these *need* to be included.

savageautomate, 2015-09-20:
I agree there is room for improvement. 

The analog input interfaces don't dictate a range or unit type.  The Pi hardware does not support any analog inputs directly, but Pi4J does support analog input when using some ADC chips via the pi4j-gpio-extensions library.

savageautomate, 2021-01-14:
**closed**; Pi4J v2 provides much better documentation in code

closed 2021-01-14 by savageautomate