Pi4J/pi4j-v1 #143: com_pi4j_wiringpi_Spi.c - Buffer size limit

Shrill-Girl, 2015-03-23:
The file com_pi4j_wiringpi_Spi.c contains a buffer size limit of 2048 bytes for all data transfers via SPI. This limit is fixed while the buffer size of the SPI driver can be changed. Also, the length parameters of the write functions are never checked against the limit, so a buffer overflow may happen during data copy - and will happen if the length argument exceeds 2048. Also the memory in the java arrays is cleared after calling the native methods.

I propose
a) to dynamically allocate buffers of the right size for the data transfer, or at least pass a user defined size limit during initialization and stay with static buffers
b) to add a flag whether the memory should be cleared or not - or additional methods which do not clear the memory

savageautomate, 2015-09-12:
I modified the native code to ```(a) to dynamically allocate buffers of the right size for the data transfer.```

closed 2015-09-12 by savageautomate

savageautomate, 2015-09-12:
This fix is now available in the latest downloadable Pi4J-1.1-SNAPSHOT build.

commit f7fed128 (2015-09-12): Fix #143; changed native buffer from fixed to to dynamically allocated memory.