Pi4J/pi4j-v1 #160: http://pi4j.com/install.html needs some additions

texadactyl, 2015-05-12:
Installed the 2015-05-05 version of Raspbian and then followed the pi4j installation instructions (preferred/easy).  All is fine except that by default, the Oracle java-related executables are not invoked.

The /usr/lib/jvm directory looks like this:
lrwxrwxrwx 1 root root   24 May  1  2012 default-java -> java-1.6.0-openjdk-armhf
lrwxrwxrwx 1 root root   20 Feb  1 02:57 java-1.6.0-openjdk-armhf -> java-6-openjdk-armhf
drwxr-xr-x 5 root root 4096 May  8 22:27 java-6-openjdk-armhf
drwxr-xr-x 3 root root 4096 May  8 22:27 java-6-openjdk-common
drwxr-xr-x 5 root root 4096 May  8 22:30 java-7-openjdk-armhf
drwxr-xr-x 9 root root 4096 May  6 18:22 jdk-8-oracle-arm-vfp-hflt

and `which java` etc. point to  java-1.6.0-openjdk-armhf.

The prospective users need to be warned that changes are necessary to make use of the Oracle 8 Java tools.  

For example, I addressed this by manually adding the following to the end of /etc/profile for subsequent logins and immediate dot(.) commands:

     export JAVA_HOME=/usr/lib/jvm/jdk-8-oracle-arm-vfp-hflt
     export PATH=$JAVA_HOME:$PATH

Then, the following compiles all of the examples with the right compiler (Oracle 8).

     . /etc/profile
     sudo bash -login /opt/pi4j/examples/build

It would be nice if the pi4j project and the Raspbian release were integrated such that Raspbian users did not have to make manual adjustments, however minor they might be.  At least, the pi4j installation document should guide folks a bit more.  Please keep in mind that many of the Raspberry Pi users while strong in an engineering field are frequently new to the Linux file system, command line, and utilities.

Thank you.

RichInTheUSA, 2015-09-12:
I believe this issue has been resolved with noobs 2015-05-11.   This latest version has java 1.8.0 and runs fine with the example programs.

texadactyl, 2015-09-14:
Thanks, Rich.  Hopefully, this resolution is present with the other Linux image downloads (i.e. w/o "noobs") such as Raspbian and Pidora.

savageautomate, 2015-09-14:
I don't recall having any install issues using Raspbian.  Oracle's Java 8 comes preinstalled and as the default JVM/JDK on the latest images from the Pi foundation.

I have not tested on Podora.

RichInTheUSA, 2015-09-14:
I did find that after installing other software on raspbian (such as eclipse), changed the default jdk  to 1.6.0.   :(  While this isn't related to pi4j, it did prevent me from being able to run the pi4j example code.

I ran "update-alternatives" for both java and javac to move both back to using 1.8.0.  Once that was done, everything ran as expected.

savageautomate, 2015-09-14:
While I have not tried it in a while, it should compile and work fine using OpenJDK 1.7 and above.  Albeit .. slower ...  I suspect the issue here is Java 1.6 as the default.  The Pi4J JARs are build with Java 1.7.

natdan, 2015-09-15:
I am not sure if this is relevant, but in case it helps: A friend of mine is stuck with old PPC MacMini but wanted to use it to develop for pi4j. RaspberryPi happily run with jdk1.8 while his computer was unable to have anything over jdk1.6. So, I have created separate branch (you'll find it in here named 1.0-jdk16) where 1.0 version of pi4j is 'downgraded' (in code) to compile and run on jdk1.6. I am not sure (can't remember) if I have published artefacts, where and under which version. The safest bet is to get source and recompile it locally.

At the moment there are no plans to do the same with any of current 1.1 snapshots not further releases. Also, I wouldn't recommend it as long as jdk1.7 or jdk1.8 can be installed on development machine.

closed 2015-09-26 by savageautomate