Pi4J/pi4j-v1 #164: I want to control GPIO on Android platform , is there any possible ?

wabp2005, 2015-06-05:
I want to control GPIO on Android platform , is there any possible ?

abhijeetramgir, 2015-06-06:
Even I want to know,
Anybody?

subnoize, 2015-06-06:
Android is Linux. If you have a rooted device with GPIO you can follow the Raspberry PI GPIO example from Oracle and write to the file handler itself, directly. There is no need for PI4J to do this as Linux is a Un*x system and everything has a file handler. You just need to know where it is and if you have permission.

--jb

savageautomate, 2015-08-06:
Pi4J itself just allows your Java code to communicate with the hardware.  You have yo add your own code to expose it over the network.

However, last year I gave a talk at Devoxx where I demonstrated using Pi4J and a Raspberry Pi to communicate sensor data via MQTT over the network to both a JavaFx application and an Android app on a mobile device. 

Check out this page for the slides and links:
http://www.savagehomeautomation.com/devoxx

The following projects are the ones I am talking about.
 - Pi4J Home Automation Sensors Demo using MQTT (Source Code + NetBeans Project)
 - Pi4J Home Automation Sensors Demo using MQTT on Android (Source Code + Android Studio Project)

closed 2015-08-06 by savageautomate