Pi4J/pi4j-v1 #171: Serial.write not working

oneenam, 2015-09-13:
I am trying to get non stop data from serial port, where I am getting data for a certain time. For example by a loop I am writing command on serial port and it works may be until 100/120 after then Serial.write is not writing any command on Serial port. Checked serial port is open or not, Its open always until I close it and also I tried by opening and closing every time to get data. Facing same problem. Tested python code where its working perfectly. When this library can't write then there is no way to get data by even python code. Need to restart the Raspberry to get data from Serial port. What could be the problem? I am assuming that because of non stop writing command the port goes on block mode...or there is no timeout  options to get data. Expecting some good solutions for this.

thanks

savageautomate, 2015-09-13:
Which version of Pi4J are you using?  If not using the 1.1-SNAPSHOT, then I would suggest giving that one a try.  The serial implementation has been completely re-written in 1.1-SNAPSHOT to be more robust and feature rich.

oneenam, 2015-09-13:
I am using "pi4j-core-1.0.jar", let me try 1.1-SNAPSHOT. thanks again for quick response.

savageautomate, 2015-09-13:
Also, the API has been changed a bit,  here is the latest example code:
https://github.com/Pi4J/pi4j/blob/develop/pi4j-example/src/main/java/SerialExample.java

And the latest Serial interfaces:
https://github.com/Pi4J/pi4j/tree/develop/pi4j-core/src/main/java/com/pi4j/io/serial

(The code for 1.1-SNAPSHOT is stored the 'develop' branch)

oneenam, 2015-09-13:
sorry for this stupid question...where is 1.1-SNAPSHOT.jar?

oneenam, 2015-09-14:
thanks. now its working much better...for one hardware by serial port is working...but if I try to get data from two hardware using serial port then its not working properly...facing same problem like previous jar....

savageautomate, 2015-09-26:
Not sure I fully understand.  Are you using a USB to serial adapter for one of the serial connections?  I think the RPi only has GPIO pins exposed for one hardware serial interface.  Thanks.

closed 2015-12-14 by savageautomate