Pi4J/pi4j-v1 #172: What is "wiringPiSetup: Unable to open /dev/mem: Operation not permitted"

yooncw, 2015-09-17:


DataWorm, 2015-09-17:
Where did this bug appear? After launching a java code that uses pi4j i guess? Have you tried to run your java code with sudo command?

yooncw, 2015-09-17:
sudo jsvc -user root

yooncw, 2015-09-17:
sudo java -jar OOO.jar
=> successed

sudo jsvc -user root 'option' OOO.jar -start
=> failed

DataWorm, 2015-09-17:
Okay that doesn't really seem to be a pi4j-related bug and i'm also not familiar with jsvc. But you could try to exec 'java -jar OOO.jar' without sudo and check whether the same error messages shows up. In this case jsvc seems to start your jar with the wrong permissions.

savageautomate, 2015-09-26:
jsvc may not be allowing the Pi4J library to run with sudo privileges.  Basically, the underlying WiringPi library can't access the /dev/mem shared memory because it does not have permissions.

closed 2015-09-26 by savageautomate

jmaerki, 2016-01-07:
I had the same problem and was able to solve it by just omitting "-user root" on jsvc. Apparently, jsvc still does some kind of downgrading when "-user root" is used.