Pi4J/pi4j-v1 #173: Extending MCP3008 to support MCP3208

DataWorm, 2015-09-17:
Pi4J already contains the MCP3008GpioProvider which works just like the MCP3208. Only difference is that MCP3208 has 12bit resolution and therefore reads 2 more values from the spi channel. Similar the MCP3204 has only 4 channels instead of 8, but that makes no difference since you can just ignore the upper 4 channels in your code. 
We could simply create a new Provider for the MCP3208 but due to code duplication an abstract class or some other approach for generalization would probably make more sense.

savageautomate, 2015-09-17:
If you look in the 'develop' branch:
https://github.com/Pi4J/pi4j/tree/develop/pi4j-gpio-extension/src/main/java/com/pi4j/gpio/extension/mcp

You may notice that I have already refactored the MCP3008GpioProvider to use an abstract base class. (https://github.com/Pi4J/pi4j/blob/develop/pi4j-gpio-extension/src/main/java/com/pi4j/gpio/extension/mcp/MCP300xGpioProvider.java) I was adding support for the MCP3004 which is exactly the same as the MCP3008 only with 4 fewer GPIOs.   

So it would seem very reasonable to add MCP3204/MCP3208 as further extensions and address the 12 bit vs 10 bit resolution in the MCP300xGpioProvider.

Want to take a crack at it?  I don't have the MCP3204/MCP3208 chips to test with.

Thanks, Robert

DataWorm, 2015-09-17:
Nope I missed that one, obviously I just watched in master branch. Anyway I just wrote the code and tested it. Seems to work fine.

savageautomate, 2015-09-17:
Great, I'll have a look at it this weekend and see if I can get it merged in.

savageautomate, 2015-09-22:
I did find some MCP3204/MCP3208 chips laying around that I did not know that I even had.  I still plan on getting this merged soon.  Work (job) keeps getting in the way :-)

DataWorm, 2015-09-24:
Don't worry, I'm student with two further jobs, I know exactly what you mean! :-D
Take your time and tell me if something needs to be improved.

savageautomate, 2015-09-26:
Merged  and available in the latest 1.1-SNAPSHOT build.  
I also added a couple examples files basically just duplicating the original MCP3004/MCP3008 examples.   Thanks!

closed 2015-09-26 by savageautomate