Pi4J/pi4j-v1 #175: GpioControllerImpl.setMode(...) throws GpioPinNotProvisionedException

Dirk1969, 2015-09-20:
Even though the pin has been provisioned before. In my opinion the bug is:

   public void setMode(PinMode mode, GpioPin... pin) {
        if (pin == null || pin.length == 0) {
            throw new IllegalArgumentException("Missing pin argument.");
        }
        for (GpioPin p : pin) {
            // ensure the requested pin has been provisioned
            if (!pins.contains(pin)) {  <<< Bug!
                throw new GpioPinNotProvisionedException(p.getPin());
            }
            // set pin mode
            p.setMode(mode);        
        }
    }

The line should be: if(!pins.contains(p)) {

Currently in test whether the whole array on pin is in the pins list, which is never the case!

savageautomate, 2015-09-20:
I think this may have been fixed in the 1.1-SNAPSHOT builds (```develop``` branch).  I'll have a look.

savageautomate, 2015-09-20:
Yep .. fixed in latest codebase:

See: https://github.com/Pi4J/pi4j/blob/4f2029ed7e49fc0cb13f1f5968b11a5c50e40154/pi4j-core/src/main/java/com/pi4j/io/gpio/impl/GpioControllerImpl.java#L168

This problem .. most likely a copy/paste error also affected other methods:
https://github.com/Pi4J/pi4j/commit/e89e4c9328d23f09d52c6a12ade8d6e93337626c

closed 2015-09-20 by savageautomate