Pi4J/pi4j-v1 #176: Trigges Does not work

NA7KR, 2015-09-25:
Does not trigger when button is pushed every time only 1 in 10 pushes.

// ***************************************
	public synchronized void GetInput1(int trigerpin, int outputpin) throws InterruptedException {
		GpioInterrupt.addListener(new GpioInterruptListener() {
			@Override
			public void pinStateChange(GpioInterruptEvent event) {
				Utils.Output_WriteDebug(true, "Raspberry Pi PIN [" + trigerpin + "] is in STATE [" + event.getState() 	+ "] Output =" + outputpin);
				try {
					GPIOController.INSTANCE.mGPIOPins[outputpin].toggle();
					mApplication.refreshGPIOPinState();
				} catch (Exception e) {
					e.printStackTrace();
				}
			}
		});
		// setup wiring pi
		if (Gpio.wiringPiSetup() == -1) {
			Utils.Output_WriteDebug(true, " ==>> GPIO SETUP FAILED");
		}
		// export all the GPIO pins that we will be using
		GpioUtil.export(trigerpin, GpioUtil.DIRECTION_IN);
		// set the edge state on the pins we will be listening for
		GpioUtil.setEdgeDetection(trigerpin, GpioUtil.DIRECTION_LOW);
		// configure GPIO x as an INPUT pin; enable it for callbacks
		Gpio.pinMode(trigerpin, Gpio.INPUT);
		Gpio.pullUpDnControl(trigerpin, Gpio.PUD_UP);
		Gpio.piHiPri(1);
		GpioInterrupt.enablePinStateChangeCallback(trigerpin);
	}
	// ***************************************

savageautomate, 2015-09-26:
If you are going to use the WiringPi methods directly like this, I would suggest not using the "```GpioInterrupt.enablePinStateChangeCallback```"  but rather take a look at the newer "```Gpio.wiringPiISR()```" method: 
https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/wiringpi/Gpio.java#L633

Here is a Java example using the  "```Gpio.wiringPiISR()```" implementation:
https://github.com/Pi4J/pi4j/blob/master/pi4j-example/src/main/java/WiringPiGpioInterruptExample2.java

"```Gpio.wiringPiISR()```" is newer and directly exposed by the WiringPi library.  
Gordon has more details on this function on the WiringPi web site: 
http://wiringpi.com/reference/priority-interrupts-and-threads/

Thanks, Robert

NA7KR, 2015-09-26:
I tried  wiringPiISR and getting error to open /sys/class/gpio/gpio18/value no such file or directory and then kills tomcat.

savageautomate, 2015-09-26:
Are you running your tomcat process with "```sudo```"?

NA7KR, 2015-09-26:
tomcat is running as root

NA7KR, 2015-09-27:
See common problem (Google) with no real fix, some downloaded from git and 
it worked and some not

NA7KR, 2015-09-28:
Fixed it with:
gpio export 18 in

echo 1 > /sys/class/gpio/gpio18/value

NA7KR, 2015-09-29:
I run sudo apt-get install raspberrypi-ui-mods and that looks to be helping.

NA7KR, 2015-10-03:
Now crash
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  Internal Error (os_linux_zero.cpp:254), pid=6552, tid=1866855520
#  fatal error: caught unhandled signal 11
#
# JRE version: OpenJDK Runtime Environment (8.0_66-b01) (build 1.8.0_66-internal-b01)
# Java VM: OpenJDK Zero VM (25.66-b01 interpreted mode linux-arm )
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

---------------  T H R E A D  ---------------

Current thread (0x017210c0):  JavaThread "Thread-4" [_thread_in_vm, id=7296, stack(0x6ec61000,0x6f460000)]

Stack: [0x6ec61000,0x6f460000],  sp=0x6f45e708,  free space=8181k

---------------  P R O C E S S  ---------------

Java Threads: ( => current thread )
=>0x017210c0 JavaThread "Thread-4" [_thread_in_vm, id=7296, stack(0x6ec61000,0x6f460000)]
  0x71cf8a88 JavaThread "http-bio-8080-exec-10" daemon [_thread_blocked, id=6582, stack(0x6fd60000,0x6fee0000)]
  0x63c19bd8 JavaThread "http-bio-8080-exec-9" daemon [_thread_in_native, id=6581, stack(0x6fee0000,0x70060000)]
  0x73407170 JavaThread "http-bio-8080-exec-8" daemon [_thread_in_native, id=6580, stack(0x70060000,0x701e0000)]
  0x74069a48 JavaThread "http-bio-8080-exec-7" daemon [_thread_in_native, id=6579, stack(0x701e0000,0x70360000)]
  0x73e3b720 JavaThread "http-bio-8080-exec-6" daemon [_thread_in_native, id=6578, stack(0x70460000,0x705e0000)]
  0x734058a0 JavaThread "http-bio-8080-exec-5" daemon [_thread_blocked, id=6577, stack(0x70818000,0x70998000)]
  0x734045a0 JavaThread "http-bio-8080-exec-4" daemon [_thread_in_native, id=6576, stack(0x70998000,0x70b18000)]
  0x734032a0 JavaThread "http-bio-8080-exec-3" daemon [_thread_in_native, id=6575, stack(0x70b18000,0x70c98000)]
  0x73402110 JavaThread "http-bio-8080-exec-2" daemon [_thread_in_native, id=6574, stack(0x70c98000,0x70e18000)]
  0x73400a98 JavaThread "http-bio-8080-exec-1" daemon [_thread_blocked, id=6573, stack(0x70e18000,0x70f98000)]
  0x72b80b20 JavaThread "http-bio-8080-AsyncTimeout" daemon [_thread_blocked, id=6572, stack(0x70f98000,0x71118000)]
  0x72b7fa90 JavaThread "http-bio-8080-Acceptor-0" daemon [_thread_in_native, id=6571, stack(0x71118000,0x71298000)]
  0x71f8a4a8 JavaThread "ContainerBackgroundProcessor[StandardEngine[Catalina]]" daemon [_thread_blocked, id=6570, stack(0x71298000,0x71418000)]
  0x71c373f0 JavaThread "GC Daemon" daemon [_thread_blocked, id=6567, stack(0x71a80000,0x71c00000)]
  0x72c3e9d8 JavaThread "Service Thread" daemon [_thread_blocked, id=6564, stack(0x720f7000,0x72277000)]
  0x72c3d058 JavaThread "Signal Dispatcher" daemon [_thread_blocked, id=6563, stack(0x72277000,0x723f7000)]
  0x72c3bd30 JavaThread "Surrogate Locker Thread (Concurrent GC)" daemon [_thread_blocked, id=6562, stack(0x723f7000,0x72577000)]
  0x72c1fe58 JavaThread "Finalizer" daemon [_thread_blocked, id=6561, stack(0x72700000,0x72880000)]
  0x72c1ea18 JavaThread "Reference Handler" daemon [_thread_blocked, id=6560, stack(0x72880000,0x72a00000)]
  0x76505f78 JavaThread "main" [_thread_in_native, id=6553, stack(0x76660000,0x767df000)]

Other Threads:
  0x72c1b880 VMThread [stack: 0x7353e000,0x735dd000] [id=6559]
  0x72c402a0 WatcherThread [stack: 0x72d47000,0x72de6000] [id=6565]

VM state:not at safepoint (normal execution)

VM Mutex/Monitor currently owned by a thread: None

NA7KR, 2015-10-03:
Do I need to send it to Java for there help is it problem with PI4J?
If I remove wiringPiISR problem goes away.

savageautomate, 2015-10-04:
Hi @NA7KR,

This native crash related to WiringPiISR() should be fixed now.
See: https://github.com/Pi4J/pi4j/issues/180

Thanks, Robert

closed 2015-10-04 by savageautomate