Pi4J/pi4j-v1 #178: When using buttons with PCF8574, the initial value is wrong

aharonha, 2015-09-30:
My configuration is I2C connected PCF8574 and 5 push buttons are connected (as digital input)
The bug is when the provider is initialized, the default pin "values" are HIGH instead of low. 
````java
        // set all default pin cache states to match documented chip power up states
        for (Pin pin : PCF8574Pin.ALL) {
            getPinCache(pin).setState(PinState.HIGH);
            currentStates.set(pin.getAddress(), true);
        }
````
Given that, when the monitor really tries to load the data for the first time - it ignores the values from the device (the buffer value is 0, meaning that the BitSet is empty - nothing to do)

````java
        public void run() {
            while (!shuttingDown) {
                try {
                    // read device pins state
                    byte[] buffer = new byte[1];
                    device.read(buffer, 0, 1);
                    BitSet pinStates = BitSet.valueOf(buffer);
                    
                    // determine if there is a pin state difference
                    for (int index = 0; index < pinStates.size(); index++) {
                        if (pinStates.get(index) != currentStates.get(index)) {
                            Pin pin = PCF8574Pin.ALL[index];
                            PinState newState = (pinStates.get(index)) ? PinState.HIGH : PinState.LOW;
````
I think that the for loop should be:
````java
for (int index = 0; index < PCF8574Pin.ALL.length; index++) {
````
or at least for the first time.

aharonha, 2015-09-30:
Actually the actual value of ````pinStates.size() ```` is 64 which can cause ````IndexOutOfBoundsException```` in line 224

savageautomate, 2015-10-04:
I am merging this pull-request today and will publish a new 1.1-SNAPSHOT build with this fix.  
Thank for for your contribution,
Robert

closed 2015-10-04 by savageautomate

aharonha, 2015-10-10:
Hi. did you merge it to the DEV branch?
I have downloaded the snapshot sources from https://oss.sonatype.org/content/repositories/snapshots/com/pi4j/pi4j-gpio-extension/1.1-SNAPSHOT/pi4j-gpio-extension-1.1-20151004.200846-21-sources.jar 

It seems that the file wasn't changed

savageautomate, 2015-10-16:
It looks like it got merged into the master branch instead of the dev branch.  Whoops. I'll merge your second pull request so that develop branch also gets this change.