Pi4J/pi4j-v1 #180: Gpio.wiringPiISR() causing native crash

savageautomate, 2015-10-04:
Pin events (callbacks) raised by the Gpio.wiringPiISR() method are resulting in a crash in in the native code.

```
<--Pi4J--> GPIO interrupt test program
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x643219dc, pid=15968, tid=1622246496
#
# JRE version: Java(TM) SE Runtime Environment (8.0-b132) (build 1.8.0-b132)
# Java VM: Java HotSpot(TM) Client VM (25.0-b70 mixed mode linux-arm )
# Problematic frame:
# C  [libpi4j176394906586883912.so+0x89dc]  cwf_7+0x18
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/pi/pi4j-dev/hs_err_pid15968.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.sun.com/bugreport/crash.jsp
#
```

savageautomate, 2015-10-04:
Fix is now available in the latest 1.1-SNAPSHOT build.
http://get.pi4j.com/download/pi4j-1.1-SNAPSHOT.deb

closed 2015-10-04 by savageautomate

NA7KR, 2015-10-09:
Sill crashes.

savageautomate, 2015-11-01:
I just tested this again today using the latest 1.1-SNAPSHOT and the included example file "WiringPiGpioInterrrupExample2" which includes several "Gpio.wiringPiISR()"s and it seems to be working and not crashing.  I am able to compile and run that example code and I am seeing GPIO events printed on screen when I tickle the physical GPIO pins.

commit ccc0df1d (2015-10-04): Fixed #180; the pointer to the callback JVM was modified a while back and the WiringPiISR() callback impl was pointing to a null instance.