Pi4J/pi4j-v1 #182: Still Crashes

NA7KR, 2015-10-09:
---------------  T H R E A D  ---------------

Current thread (0x61dacb88):  JavaThread "Thread-6" [_thread_in_vm, id=16846, stack(0x5cd01000,0x5d500000)]

Stack: [0x5cd01000,0x5d500000],  sp=0x5d4fe708,  free space=8181k

---------------  P R O C E S S  ---------------

Java Threads: ( => current thread )
  0x61db4b48 JavaThread "Thread-5" [_thread_in_vm, id=14835, stack(0x5e001000,0x5e800000)]
=>0x61dacb88 JavaThread "Thread-6" [_thread_in_vm, id=16846, stack(0x5cd01000,0x5d500000)]
  0x61c773d8 JavaThread "Atmosphere-Scheduler-56" daemon [_thread_blocked, id=17062, stack(0x60080000,0x60200000)]
  0x62a386c0 JavaThread "Atmosphere-Scheduler-5353" daemon [_thread_blocked, id=17061, stack(0x5f380000,0x5f500000)]
  0x73e14770 JavaThread "Atmosphere-Scheduler-5355" daemon [_thread_blocked, id=17060, stack(0x5fa80000,0x5fc00000)]
  0x61c87e30 JavaThread "Atmosphere-Scheduler-0" daemon [_thread_blocked, id=16834, stack(0x5fd80000,0x5ff00000)]
  0x62a58248 JavaThread "http-bio-8080-exec-40" daemon [_thread_blocked, id=16455, stack(0x60d57000,0x60ed7000)]
  0x0125a0c8 JavaThread "http-bio-8080-exec-39" daemon [_thread_blocked, id=16452, stack(0x60957000,0x60ad7000)]
  0x6069b520 JavaThread "http-bio-8080-exec-38" daemon [_thread_blocked, id=16449, stack(0x5f600000,0x5f780000)]
  0x01259c28 JavaThread "http-bio-8080-exec-37" daemon [_thread_blocked, id=16448, stack(0x5fc00000,0x5fd80000)]
  0x6068d5e8 JavaThread "http-bio-8080-exec-36" daemon [_thread_blocked, id=16445, stack(0x5ff00000,0x60080000)]
  0x7403ffe0 JavaThread "http-bio-8080-exec-35" daemon [_thread_blocked, id=16440, stack(0x61700000,0x61880000)]
  0x61c7fe20 JavaThread "http-bio-8080-exec-34" daemon [_thread_blocked, id=16435, stack(0x612e1000,0x61461000)]
  0x01261a18 JavaThread "http-bio-8080-exec-33" daemon [_thread_blocked, id=16432, stack(0x5f200000,0x5f380000)]
  0x7421f948 JavaThread "http-bio-8080-exec-32" daemon [_thread_blocked, id=16429, stack(0x61580000,0x61700000)]
  0x74212690 JavaThread "http-bio-8080-exec-31" daemon [_thread_blocked, id=16428, stack(0x61057000,0x611d7000)]
  0x61c84560 JavaThread "Atmosphere-Scheduler-5111" daemon [_thread_blocked, id=16238, stack(0x5f900000,0x5fa80000)]
  0x73e0c8e8 JavaThread "Atmosphere-Scheduler-3751" daemon [_thread_blocked, id=11939, stack(0x60300000,0x60480000)]
  0x61f80448 JavaThread "http-bio-8080-AsyncTimeout" daemon [_thread_blocked, id=576, stack(0x607d7000,0x60957000)]
  0x61f800d8 JavaThread "http-bio-8080-Acceptor-0" daemon [_thread_in_native, id=575, stack(0x60ed7000,0x61057000)]
  0x61f7faf8 JavaThread "ContainerBackgroundProcessor[StandardEngine[Catalina]]" daemon [_thread_blocked, id=574, stack(0x60ad7000,0x60c57000)]
  0x61c37540 JavaThread "GC Daemon" daemon [_thread_blocked, id=518, stack(0x61980000,0x61b00000)]
  0x62c3eb20 JavaThread "Service Thread" daemon [_thread_blocked, id=500, stack(0x620f7000,0x62277000)]
  0x62c3d1a0 JavaThread "Signal Dispatcher" daemon [_thread_blocked, id=499, stack(0x62277000,0x623f7000)]
  0x62c3be78 JavaThread "Surrogate Locker Thread (Concurrent GC)" daemon [_thread_blocked, id=498, stack(0x623f7000,0x62577000)]
  0x62c1fe58 JavaThread "Finalizer" daemon [_thread_blocked, id=494, stack(0x62700000,0x62880000)]
  0x62c1ea18 JavaThread "Reference Handler" daemon [_thread_blocked, id=493, stack(0x62880000,0x62a00000)]
  0x76505f78 JavaThread "main" [_thread_in_native, id=467, stack(0x766cc000,0x7684b000)]

Other Threads:
  0x62c1b880 VMThread [stack: 0x62d47000,0x62de6000] [id=492]
  0x62c403e8 WatcherThread [stack: 0x62b61000,0x62c00000] [id=501]

VM state:not at safepoint (normal execution)

NA7KR, 2015-10-09:
pi4j-core-1.1-SNAPSHOT.jar

NA7KR, 2015-10-14:
does this help any if you need the full file I can email it to you.

<a 'java/net/SocketTimeoutException': Read timed out> (0x641ff588) thrown at [/build/openjdk-8-2oJtqp/openjdk-8-8u66-b01/src/hotspot/src/share/vm/prims/jni.cpp, line 735]
Event: 184.226 Thread 0x63402590 Exception <a 'java/net/SocketTimeoutException': Read timed out> (0x641ff830) thrown at [/build/openjdk-8-2oJtqp/openjdk-8-8u66-b01/src/hotspot/src/share/vm/prims/jni.cpp, line 735]
Event: 184.227 Thread 0x002ef900 Exception <a 'java/net/SocketTimeoutException': Read timed out> (0x641ffbf0) thrown at [/build/openjdk-8-2oJtqp/openjdk-8-8u66-b01/src/hotspot/src/share/vm/prims/jni.cpp, line 735]

NA7KR, 2015-10-18:
Any news when you can have a look at this problem?

NA7KR, 2015-10-22:
Do not know where the problem is so opened a bug with Java also JI-9025646: Fairoz Matte at Oracle has the case and looking in to it.

NA7KR, 2015-10-23:
The works OK from cmd line.
I crashes every time in Tomcat 7 and 8
Please can you have  a look at error. 
		}
		Gpio.pinMode(trigerpin, Gpio.INPUT);
		Gpio.pullUpDnControl(trigerpin, Gpio.PUD_UP);
		Gpio.wiringPiISR(trigerpin, Gpio.INT_EDGE_FALLING, new GpioInterruptCallback() {
			@Override
			public void callback(int pin) {
				System.out.println(" ==>> GPIO PIN " + pin + " - INTERRUPT DETECTED");	
			}
		});

savageautomate, 2015-10-30:
Sorry, I have been traveling for a few weeks and have not been able to look at this.  So from your last comment this code example is working outside of Tomcat, but crashes when run with Tomcat.  You may have answered this in a previous thread, so forgive me if I don't recall, but are you launching Tomcat with root/sudo privileges?  

Thanks, Robert

NA7KR, 2015-10-30:
Yes running as root

NA7KR, 2015-10-30:
if you want view my code it on https://github.com/NA7KR/YT-100_ATU_Controller.git

NA7KR, 2015-10-30:
I tested with 8.51, 8.60, 8.65 and 8.72 and all have problems and Tomca7 and 8

savageautomate, 2015-11-01:
OK, I took a look at your code and I wonder if the issue is starting with this line:
https://github.com/NA7KR/YT-100_ATU_Controller/blob/master/src/main/java/com/na7kr/Interrupt/Interrupt.java#L28

Your callback passed into "Gpio.wiringPiISR" is defined inline as an anonymous inner function.  However, after the call to "public synchronized void GetInput1()" is complete the anonymous inner callback function goes out of scope and gets destroyed.  I suspect that the native code is trying to invoke a callback on a Java function that is not longer available?  What do you think about this theory?  

Ideally I should probably add some code to prevent am abrupt crash :-) 

Thanks, Robert

savageautomate, 2015-11-01:
Yes, a complete copy of the dumpfile/stacktrace would be helpful.  Looking specifically for the last filename and line number that reference some Pi4J code before the crash.

NA7KR, 2015-11-01:
Does not show line number or any that help but you can have a look... do you have a place to upload?

savageautomate, 2015-11-01:
You can email it to me:  robert at pi4j.com

NA7KR, 2015-12-03:
Killed project as does not work with tomcat

commit a59064aa (2015-11-01): Attempting to fix #182 by preventing JNI crash on a Java callback interface that is no longer available at callback invocation time.
closed 2015-12-03 by NA7KR