Pi4J/pi4j-v1 #183: MCP23008GpioProvider. - java.lang.ArrayIndexOutOfBoundsException: 128

GeorgeProost, 2015-10-09:
Hi, Thanks for the  great wrapper.

I have located a possible addressing issue in MCP23008GpioProvider or thereabouts. Cannot seem to address all the gpio pins on the MCP23008 chip. GPIO_07 (ie 8th one).

Any ideas?

Regards 

George

GeorgeProost, 2015-10-10:
Hi,

I think the issue lies deeper, not in pi4j ...

Regards

George

GeorgeProost, 2015-10-11:
Hi, I am reopening the issue, does seem to have to do with pin caching.
If there is any more info required, I can try to get it to you. I include the source:
package sfcountertester;

import com.pi4j.gpio.extension.mcp.MCP23008GpioProvider;
import com.pi4j.gpio.extension.mcp.MCP23008Pin;
import com.pi4j.io.gpio.GpioController;
import com.pi4j.io.gpio.GpioFactory;
import com.pi4j.io.gpio.GpioPinDigitalOutput;
import com.pi4j.io.gpio.PinState;
import com.pi4j.io.i2c.I2CBus;
import java.io.IOException;

/**
 *
 * @author gproost
 */
public class SFCounterTester implements Runnable {

    private int numRelays;
    private int minimumClicks = 10;
    private int delay = 100;

    // create gpio controller
    final GpioController gpio = GpioFactory.getInstance();

    // create custom MCP23008 GPIO provider
    final MCP23008GpioProvider gpioProvider;

    // relays
    static GpioPinDigitalOutput relay1;
    static GpioPinDigitalOutput relay2;
    static GpioPinDigitalOutput relay3;
    static GpioPinDigitalOutput relay4;
    static GpioPinDigitalOutput relay5;
    static GpioPinDigitalOutput relay6;
    static GpioPinDigitalOutput relay7;
    static GpioPinDigitalOutput relay8;

    /**
     * * *
     *
     * @param numRelays
     * @param delay
     * @throws InterruptedException
     * @throws java.io.IOException
     */
    public SFCounterTester(int numRelays, int delay) throws InterruptedException, IOException {

        gpioProvider = new MCP23008GpioProvider(I2CBus.BUS_1, 0x20);

        try {

            this.numRelays = numRelays;
            this.delay = delay;

            if (numRelays >= 1) {
                relay1 = gpio.provisionDigitalOutputPin(gpioProvider, MCP23008Pin.GPIO_00, "Relay8", PinState.LOW);
                relay1.setShutdownOptions(true, PinState.LOW);
            }

            if (numRelays >= 2) {
                relay2 = gpio.provisionDigitalOutputPin(gpioProvider, MCP23008Pin.GPIO_01, "Relay7", PinState.LOW);
                relay2.setShutdownOptions(true, PinState.LOW);
            }

            if (numRelays >= 3) {
                relay3 = gpio.provisionDigitalOutputPin(gpioProvider, MCP23008Pin.GPIO_02, "Relay6", PinState.LOW);
                relay3.setShutdownOptions(true, PinState.LOW);

            }

            if (numRelays >= 4) {
                relay4 = gpio.provisionDigitalOutputPin(gpioProvider, MCP23008Pin.GPIO_03, "Relay5", PinState.LOW);
                relay4.setShutdownOptions(true, PinState.LOW);
            }

            if (numRelays >= 5) {
                relay5 = gpio.provisionDigitalOutputPin(gpioProvider, MCP23008Pin.GPIO_04, "Relay4", PinState.LOW);
                relay5.setShutdownOptions(true, PinState.LOW);

            }

            if (numRelays >= 6) {
                relay6 = gpio.provisionDigitalOutputPin(gpioProvider, MCP23008Pin.GPIO_05, "Relay3", PinState.LOW);
                relay6.setShutdownOptions(true, PinState.LOW);
            }

            if (numRelays >= 7) {
                relay7 = gpio.provisionDigitalOutputPin(gpioProvider, MCP23008Pin.GPIO_06, "Relay2", PinState.LOW);
                relay7.setShutdownOptions(true, PinState.LOW);
            }

            if (numRelays == 8) {
                relay8 = gpio.provisionDigitalOutputPin(gpioProvider, MCP23008Pin.GPIO_07, "Relay1", PinState.LOW);
                relay8.setShutdownOptions(true, PinState.LOW);
            }

        } catch (Exception ex) {
            System.out.println("A serious error has occured:");
            ex.printStackTrace();
        }
    }

    @Override
    public void run() {

        switch (numRelays) {

            case 8:
                runTest(relay8, minimumClicks + 1, this.delay);

            case 7:
                runTest(relay7, minimumClicks + 2, this.delay);

            case 6:
                runTest(relay6, minimumClicks + 3, this.delay);

            case 5:
                runTest(relay5, minimumClicks + 4, this.delay);

            case 4:
                runTest(relay4, minimumClicks + 5, this.delay);

            case 3:
                runTest(relay3, minimumClicks + 6, this.delay);

            case 2:
                runTest(relay2, minimumClicks + 7, this.delay);

            case 1:
                runTest(relay1, minimumClicks + 8, this.delay);

        }

        // get out gracefully
        gpioProvider.shutdown();
        //       gpio.shutdown();

    }

    /**
     *
     * @param pin
     * @param number
     * @throws InterruptedException
     */
    private void runTest(GpioPinDigitalOutput pin, int number, int delay) {

        try {

            System.out.println("Expect: " + number + " counts for relay: " + pin.getName());

            for (int i = 0; i <= number - 1; i++) {

                gpio.setState(PinState.HIGH, pin);
                Thread.sleep(delay);

                gpio.setState(PinState.LOW, pin);
                Thread.sleep(delay);
            }

        } catch (Exception e) {

        }
    }
}

GeorgeProost, 2015-10-11:
Hi,

I have dug a bit deeper:

1. I am using snapshot-1.0.1 .. I get the error
2. reverted to v 1.0 ... and it works

I do however need the snaphot version for serial.

hope this helps,

George

mihainigrini, 2015-10-12:
Hi George
\snapshot 1.1
Ty
Mihai

GeorgeProost, 2015-10-12:
Hi Mihai,

Serial is totally changed in snapshot, a complete new implementation and works great.
The serial functionality in release 1.0 is very basic and lacks some configuration options, so yes 
for serial you will need to change some things quite drastically if you flip flop between them.

My concern is with the MCP23008 implementation.  Its goes from working in 1.0 to broken in snapshot.
As snapshot has the serial functionality I need I switched to that, but breaks the MCP23008.

Seems an addressing exception of some sort, but on further investigation could be similar to a previous issue when an int is cast to byte and the high order bit does not flip, so addressing breaks and we cannot reference 128.

I have not worked github before so am still reluctant to branch/debug/fix/commit.

I am hoping that Mr. Savage or another developer can rectify this issue.

I am still in 2 minds about reimplementing v1.0 and use another library for serial. I would rather just use pi4j.

Hope this helps

George

savageautomate, 2015-10-16:
I would rather just get it fixed in the 1.1-SNAPSHOT builds too.  There were some recent optimizations done for pi caching, I suspect that that may have introduced the issue here.  Not sure will have to dig in.

GeorgeProost, 2015-10-31:
Thanks, appreciate your efforts.

RasPelikan, 2015-12-20:
Hello,

I got the same error using MCP23017:

SEVERE: Error running application
java.lang.ArrayIndexOutOfBoundsException: 128
	at com.pi4j.io.gpio.GpioProviderBase.getPinCache(GpioProviderBase.java:69)
	at com.pi4j.io.gpio.GpioProviderBase.setState(GpioProviderBase.java:179)
	at com.pi4j.gpio.extension.mcp.MCP23017GpioProvider.setState(MCP23017GpioProvider.java:234)
....

Any news around this issue?

Thanks,
Stephan

RasPelikan, 2015-12-20:
In MCP23017Pin.java shows

public class MCP23017Pin {
    
    public static final Pin GPIO_A0 = createDigitalPin(1, "GPIO A0");
    public static final Pin GPIO_A1 = createDigitalPin(2, "GPIO A1");
    public static final Pin GPIO_A2 = createDigitalPin(4, "GPIO A2");
    public static final Pin GPIO_A3 = createDigitalPin(8, "GPIO A3");
    public static final Pin GPIO_A4 = createDigitalPin(16, "GPIO A4");
    public static final Pin GPIO_A5 = createDigitalPin(32, "GPIO A5");
    public static final Pin GPIO_A6 = createDigitalPin(64, "GPIO A6");
    public static final Pin GPIO_A7 = createDigitalPin(128, "GPIO A7");
    public static final Pin GPIO_B0 = createDigitalPin(1001, "GPIO B0");
    public static final Pin GPIO_B1 = createDigitalPin(1002, "GPIO B1");
    public static final Pin GPIO_B2 = createDigitalPin(1004, "GPIO B2");
    public static final Pin GPIO_B3 = createDigitalPin(1008, "GPIO B3");
    public static final Pin GPIO_B4 = createDigitalPin(1016, "GPIO B4");
    public static final Pin GPIO_B5 = createDigitalPin(1032, "GPIO B5");
    public static final Pin GPIO_B6 = createDigitalPin(1064, "GPIO B6");
    public static final Pin GPIO_B7 = createDigitalPin(1128, "GPIO B7");

I use GPIO_A7 which is initialized by address 128. 

GpioProviderBase.java got

    protected GpioProviderPinCache getPinCache(Pin pin) {
        GpioProviderPinCache pc = cache[pin.getAddress()];

so the address (im my case 128) is used as cache-index but in the same class the cache is initialized by

    protected GpioProviderPinCache[] cache = new GpioProviderPinCache[100]; // support up to pin address 100

This was change in commit 8134a6f back in September. In seems that the term "address" is used in a different manner. It seems that address in the cache is used as a number from 0 to 40 according the gpio-header of the Raspberry Pi. In MCP23017Pin the address is the address used in I2C communication for identifying the pin. I will try to separate them...

RasPelikan, 2015-12-20:
https://github.com/Pi4J/pi4j/pull/200

savageautomate, 2016-03-15:
OK, I think we have fixed the underlying issue with the pin cache array capacity.  You can see my comments in https://github.com/Pi4J/pi4j/issues/224 for the details.

closed 2016-03-15 by savageautomate