Pi4J/pi4j-v1 #185: W1: ds18b20 not working anymore

DataWorm, 2015-10-14:
Hi,

with the new w1-support in develop branch I got a ds18b20 temperature sensor running. Currently that device type does no longer get recognized. Running W1TempExample.java from pi4j-example prints following lines: 
INFO: no device type for [28] found - ignoring
W1 Master: /sys/bus/w1/devices
Device Types: 
 - fe = class com.pi4j.io.w1.W1DummyDeviceType$W1DummyDevice
Devices:

I'm not quite sure why that type does no longer get recognized. In my project I used pi4j-distribution as maven dependency but when I tried to compile it today (it also worked before, but it's a while ago), I just received errors that there is no jar found (obviously because it's a deb package). Maybe those sudden appearances are related to each other.
What could be the reason that this device type is no longer found?

schup, 2015-10-14:
Are there any devices present?

What is the output of 

    ls -la  /sys/bus/w1/devices

DataWorm, 2015-10-14:
Yep, the message "INFO: no device type for [28] found - ignoring" already indicates that the device was found. It just fails to recognize '28' as a device type for temperature sensors because that device type is not registered (also visible in the output that only shows the dummy device type). I am not familiar with those service class loaders and how those device types get registered. I have to read more about that to get an idea what the problem could be or where to search for it. I just thought maybe the device types could be defined/registered in  pi4j-distribution?!

schup, 2015-10-14:
Ah, yes, I should have noticed.

The implemenation for the ds18b20 is in the pi4j-device jar.
All that is required is that this jar is in the classpath.

ServiceLoader scans the classpath for META-INF/services/com.pi4j.io.w1.W1DeviceType files.
This file simply lists implemenations for the com.pi4j.io.w1.W1DeviceType interface.

How does your classpath look like?

DataWorm, 2015-10-14:
Ah thanks for your help, this bug is obviously not related to pi4j. I was testing the temperature example with a remote tool (LaunchPi) from eclipse. In there seem to be the first issue but it's not important because I just use it for tests. 
The same issue happened in a driver that I tried to build with pi4j for temperature sensors. Those drivers are deployed as jar files and will be loaded by the framework during runtime. This driver has it's own classloader for sure and maybe there is something wrong...

DataWorm, 2015-10-14:
Yep, it was just as I have expected! The service loader is called without specifying a classloader, therefore it just picks the classloader of the current thread. In case of the framework I used, the thread seems to belong to the classloader of the framework instead of the driver. 

The service loader gets already called by constructor of W1Master. Therefore I think we could just create a second constructor which accepts a user-defined classloader. Do you have any other suggestion?!

schup, 2015-10-15:
Adding a constructor with a classloader and adding the ClassLoader to the the init method is a good approach.

It should be safe to call the init with a classloader multiple times but that's not tested :-)

natdan, 2015-10-29:
@DataWorm , @schup - to me it seems that something else is fundamentally wrong - not missing constructor with a classloader. 
What 'framework' are we talking about here? I think that if your project is started up correctly (with all jar files supplied) it should just work (current thread's classloader would have access to implementation jar)...

@DataWorm BTW there's an alternative to LaunchPI, maybe not that sophisticated with UI, but works seamlessly: http://remotevm.abstracthorizon.org/ 
I don't mind giving you a hand to set it up if you have any issues. Maybe using that one would sort out your immediate classloading problems...

DataWorm, 2015-10-29:
Thanks for the link, set a bookmark on it and will try it out on occasion.

Framework might be the wrong word. I was talking about IOLITE, a smart home platform which supports hot-deployable drivers and apps. Therefore it has to load external jars at runtime with a seperate classloader.

natdan, 2015-10-29:
Do you have any link I can learn more about it? I am just curious if your only problem is pi4j stuff you are using being in several different jars... Worse case scenario you can always re-pack them into one jar and you won't have any issues... But, I am talking from 'feeling' what the issue might be rather than from perspective of facts. O: )

DataWorm, 2015-10-29:
Yep, I could somehow provide a link but the software is not published yet (I am one of the developers behind that software). Maybe you could get access to a developer preview, but I need to discuss that with my boss first. Some general infos about IOLITE are available on this page: http://iolite.de/
For further details, you can also contact me by mail (motza $at$ dai-labor.de).

IOLITE is based on java and I use it on my raspberry. I wrote a driver so that the platform does recognize DS18B20 devices as temperature sensors and integrates them. This driver just uses that w1 implementations and periodically checks for new devices or new values of already found devices. The driver is build into a single jar file with all dependencies included. That jar-file gets loaded from IOLITE at runtime and initiated by calling some lifecycle methods which are pre-defined by an interface.Within that lifecycle I start to create an instance of W1Master to start looking for new devices (which results in the classloader call). That lifecycle methods get of course called by IOLITE. 

Because of hot-deployment and security every driver/app is started with  a seperated classloader. But the lifecycle methods seem to be called by a thread belonging to the IOLITE platform and therefore the corresponding classloader of that thread is also mapped to IOLITE instead of my driver. The result is that the serviceloader does just look into the classloader of iolite instead of my driver which includes the pi4j-deps. Therefore I made that quick fix so that I was able to use the classloader of my driver class instead of the running thread.

DataWorm, 2018-07-01:
I guess this ticket can be closed =)

closed 2018-07-01 by DataWorm