Pi4J/pi4j-v1 #189: GPIO Performance is slow due to too many calls to GpioUtil.isPinSupported

lff0305, 2015-11-08:
I wrote a small test class to see how fast p4j can toggle the GPIO state:

```java
import com.pi4j.io.gpio.GpioController;
import com.pi4j.io.gpio.GpioFactory;
import com.pi4j.io.gpio.GpioPinDigitalOutput;
import com.pi4j.io.gpio.PinState;
import com.pi4j.io.gpio.RaspiPin;

public class TestSpeed {
	private static GpioPinDigitalOutput DIO;

	static final GpioController gpio = GpioFactory.getInstance();

	public static void main(String[] argu) {
		init();
		test();
		gpio.shutdown();
	}

	private static void test() {
		long l0 = System.currentTimeMillis();
		int count = 10000;
		for (int i = 0; i < count; i++) {
			DIO.high();
			DIO.low();
		}
		long time = System.currentTimeMillis() - l0;
		System.out.println(time + " used.");

		float per = time / (float) count;

		System.out
				.println(per + " per toggle, " + (1000 / per) + " per second");
	}

	private static void init() {
		// create gpio controller
		DIO = gpio.provisionDigitalOutputPin(RaspiPin.GPIO_04, "dio",
				PinState.HIGH);
	}
}
```

But the performance is not good. After some investigation, I found that GPIO.isPinSupported is called every time high() or low() is called (I add some code to test)

        at com.pi4j.wiringpi.GpioUtil.isPinSupported(GpioUtil.java:291)
        at com.pi4j.io.gpio.RaspiGpioProvider.hasPin(RaspiGpioProvider.java:61)
        at com.pi4j.io.gpio.GpioProviderBase.setState(GpioProviderBase.java:172)
        at com.pi4j.io.gpio.RaspiGpioProvider.setState(RaspiGpioProvider.java:154)
        at com.pi4j.io.gpio.impl.GpioPinImpl.setState(GpioPinImpl.java:325)
        at com.pi4j.io.gpio.impl.GpioPinImpl.low(GpioPinImpl.java:209)
        at TestSpeed.test(TestSpeed.java:23)
        at TestSpeed.main(TestSpeed.java:14)

So I think there should be a way to cache it (Do not call every time)

lff0305, 2015-11-08:
I changed the code a little, to 
```java
    private static ConcurrentHashMap<Pin, Boolean> hasPinMap = new ConcurrentHashMap();
    
    @Override
    public boolean hasPin(Pin pin) {
    	Boolean r = hasPinMap.get(pin);
    	if (r != null) {
    		return r.booleanValue();
    	}
        boolean realValue = (com.pi4j.wiringpi.GpioUtil.isPinSupported(pin.getAddress()) > 0);
        hasPinMap.putIfAbsent(pin, realValue);
        return realValue;
    }
```
And the test result is much better (from 1400 per second to 158730 per second). I do not know if my code is reasonable, but it seems to be working...

savageautomate, 2015-11-08:
Have you tried using the 1.!-SNAPSHOT build?  (located in 'develop' branch)  
I think some work has already been done in this area to improve performance.

lff0305, 2015-11-08:
You are correct. 1.1 snapshot fixed this bug!

closed 2015-11-08 by lff0305