Pi4J/pi4j-v1 #190: stack trace

NA7KR, 2015-11-12:
Why do I get this?

Kevin

root@pi:~# Nov 11, 2015 8:45:37 PM org.apache.catalina.loader.WebappClassLoaderBase checkStateForResourceLoading
INFO: Illegal access: this web application instance has been stopped already. Could not load [com.pi4j.wiringpi.GpioInterruptEvent]. The following stack trace is thrown for debugging purposes as well as to attempt to terminate the thread which caused the illegal access.
java.lang.IllegalStateException: Illegal access: this web application instance has been stopped already. Could not load [com.pi4j.wiringpi.GpioInterruptEvent]. The following stack trace is thrown for debugging purposes as well as to attempt to terminate the thread which caused the illegal access.
        at org.apache.catalina.loader.WebappClassLoaderBase.checkStateForResourceLoading(WebappClassLoaderBase.java:1355)
        at org.apache.catalina.loader.WebappClassLoaderBase.checkStateForClassLoading(WebappClassLoaderBase.java:1341)
        at org.apache.catalina.loader.WebappClassLoaderBase.loadClass(WebappClassLoaderBase.java:1206)
        at org.apache.catalina.loader.WebappClassLoaderBase.loadClass(WebappClassLoaderBase.java:1167)
        at com.pi4j.wiringpi.GpioInterrupt.pinStateChangeCallback(GpioInterrupt.java:127)

[DEBUG] 2015-11-11-20:45:37 [Thread-59] InterruptListener.ReadInterrupt (ReadInterrupt.java:20) - Raspberry Pi PIN [1] is in STATE [true]
Exception in thread "Thread-58" java.lang.IllegalStateException: Can't overwrite cause with java.lang.IllegalStateException: Illegal access: this web application instance has been stopped already. Could not load [com.pi4j.wiringpi.GpioInterruptEvent]. The following stack trace is thrown for debugging purposes as well as to attempt to terminate the thread which caused the illegal access.
        at java.lang.Throwable.initCause(Throwable.java:457)
        at org.apache.catalina.loader.WebappClassLoaderBase.checkStateForClassLoading(WebappClassLoaderBase.java:1344)
        at org.apache.catalina.loader.WebappClassLoaderBase.loadClass(WebappClassLoaderBase.java:1206)
        at org.apache.catalina.loader.WebappClassLoaderBase.loadClass(WebappClassLoaderBase.java:1167)
        at com.pi4j.wiringpi.GpioInterrupt.pinStateChangeCallback(GpioInterrupt.java:127)
Caused by: java.lang.ClassNotFoundException
        at org.apache.catalina.loader.WebappClassLoaderBase.checkStateForClassLoading(WebappClassLoaderBase.java:1343)
        ... 3 more
[DEBUG] 2015-11-11-20:45:37 [Thread-60] InterruptListener.ReadInterrupt (ReadInterrupt.java:20) - Raspberry Pi PIN [1] is in STATE [false]
Nov 11, 2015 8:45:37 PM org.apache.catalina.loader.WebappClassLoaderBase checkStateForResourceLoading
INFO: Illegal access: this web application instance has been stopped already. Could not load [com.pi4j.wiringpi.GpioInterruptEvent]. The following stack trace is thrown for debugging purposes as well as to attempt to terminate the thread which caused the illegal access.
java.lang.IllegalStateException: Illegal access: this web application instance has been stopped already. Could not load [com.pi4j.wiringpi.GpioInterruptEvent]. The following stack trace is thrown for debugging purposes as well as to attempt to terminate the thread which caused the illegal access.
        at org.apache.catalina.loader.WebappClassLoaderBase.checkStateForResourceLoading(WebappClassLoaderBase.java:1355)
        at org.apache.catalina.loader.WebappClassLoaderBase.checkStateForClassLoading(WebappClassLoaderBase.java:1341)
        at org.apache.catalina.loader.WebappClassLoaderBase.loadClass(WebappClassLoaderBase.java:1206)
        at org.apache.catalina.loader.WebappClassLoaderBase.loadClass(WebappClassLoaderBase.java:1167)
        at com.pi4j.wiringpi.GpioInterrupt.pinStateChangeCallback(GpioInterrupt.java:127)

Exception in thread "Thread-61" java.lang.IllegalStateException: Can't overwrite cause with java.lang.IllegalStateException: Illegal access: this web application instance has been stopped already. Could not load [com.pi4j.wiringpi.GpioInterruptEvent]. The following stack trace is thrown for debugging purposes as well as to attempt to terminate the thread which caused the illegal access.
        at java.lang.Throwable.initCause(Throwable.java:457)
        at org.apache.catalina.loader.WebappClassLoaderBase.checkStateForClassLoading(WebappClassLoaderBase.java:1344)
        at org.apache.catalina.loader.WebappClassLoaderBase.loadClass(WebappClassLoaderBase.java:1206)
        at org.apache.catalina.loader.WebappClassLoaderBase.loadClass(WebappClassLoaderBase.java:1167)
        at com.pi4j.wiringpi.GpioInterrupt.pinStateChangeCallback(GpioInterrupt.java:127)
Caused by: java.lang.ClassNotFoundException
        at org.apache.catalina.loader.WebappClassLoaderBase.checkStateForClassLoading(WebappClassLoaderBase.java:1343)
        ... 3 more

NA7KR, 2015-11-20:
Is there a better way with out tomcat?

NA7KR, 2015-12-03:
I killed my project as PI4J does not work with tomcat.

closed 2015-12-03 by NA7KR

peyrona, 2016-12-20:
Hello.
I've found exactly the same problem.
Did you solved? Is there any workaround?
Thank you very much in advance.

rlsutton1, 2016-12-20:
I use pi4j with jetty. Here is a link to my class which starts jetty https://github.com/rlsutton1/Thermostat/blob/master/src/main/java/au/com/rsutton/entryPoint/Main.java

peyrona, 2016-12-21:
> main/java/au/com/rsutton/entryPoint/Main.java

Hello again.

Thanks a million for answering so quickly.

Best regards.
--------------------
Francisco Morero Peyrona
www.peyrona.com
--------------------