Pi4J/pi4j-v1 #191: ArithmeticException: Non-terminating decimal expansion in PCA9685GpioServoDriver

cjm27, 2015-11-15:
Exception in thread "main" java.lang.ArithmeticException: Non-terminating decimal expansion; no exact representable decimal result.
        at java.math.BigDecimal.divide(BigDecimal.java:1690)
        at com.pi4j.component.servo.impl.PCA9685GpioServoDriver.updateResolution(PCA9685GpioServoDriver.java:87)

Problem should be able to be solved by adding a rounding parameter in the divide method, like BigDecimal.ROUND_HALF_DOWN

savageautomate, 2015-11-15:
Closed.  Pull request merged: https://github.com/Pi4J/pi4j/commit/df26e4da7a5b50fed8ea3bc04043b80f5eb7490a

closed 2015-11-15 by savageautomate

freeloki, 2016-07-17:
i've installed the last version of pi4j for my rpi3 but i'm still getting this error on "PCA9685GpioServoExample" 

how can i fix this issue ? Thanks.

rlsutton1, 2016-07-17:
This problem keeps coming up...

Why is com.pi4j.component.servo.impl.PCA9685GpioServoDriver.updateResolution(...) using BigDecimal? this is after all just servo positioning not a banking transaction! I think an error of 10^-16 is acceptable.

Just change it to use Double like this...

resolution = (int) (4096.0/getProvider().getFrequency());

freeloki, 2016-07-17:
resolution = (int) (4096.0/getProvider().getFrequency()); // this solution's giving build error but i fixed it  by changing :

resolution = (int) (4096.0/getProvider().getFrequency()**.intValueExact()**);

Thanks for reply.

natdan, 2016-07-18:
Maybe you can set up an a pull request with those changes? :)