Pi4J/pi4j-v1 #196: i2c receiving more 100 bytes

lettierimarco, 2015-12-07:
Hi, i've some trouble coomunicating raspberry-arduino via i2c.

Arduino Sketch

#include <Wire.h>
void setup() {
  Wire.begin(9); 
  Wire.onRequest(requestEvent);
}
void loop() {
  delay(100);
}
void requestEvent() {
 Wire.write("vf9xk0ejicwf90ljqt00c5kdk8hynbdowdc63wvsyhxyno737da9o0syyz2rh66jpogya1wfpjj2ywcbzcrd29h6fan2mzbhdhlup8fkxoc1au3fckykz5gklacxzekj1wjwuo6hip9ipi3s8r2xmpoy5q1aaszie8ke2xlac0y787p6oyahtoloks41au9u8zyhmqx4gm2vzh7elnl4hwy8utzvtef734eg02a6u27jqlnhyec6gg41f92yd7wm"); }

Raspberry Java

I2CBus i2c_bus = I2CFactory.getInstance(I2CBus.BUS_1);
I2CDevice avr_device = i2c_bus.getDevice(9);
avr_device.read(packet,0,256);

The output in packet is 0x00 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF...

natdan, 2015-12-07:
I'll return back to you regarding this. But it might be slightly later as we are heading for the last few weeks before Christmas.

Anyway - just for the reference - I have made some assembly code for ATtiny 2313 that uses i2c and it worked without any issues with RPi. Unfortunately I didn't use Arduino's i2c library for it so I not 100% sure what could be wrong in your example. Double check all the timing as they can cause issues...

colinrgodsey, 2016-08-03:
I found an issue when read/write buffers larger than 127 bytes, going to have a PR for it soon

savageautomate, 2016-12-02:
What this issue resolved?

closed 2016-12-02 by lettierimarco