Pi4J/pi4j-v1 #197: SPI size limit

empirephoenix, 2015-12-10:
Hi,

is there a reason for the hardcoded limit of the buffer in the SPI..c class for the buffer?
In my opinion it should be possible to instead determine the necessary size with the information from the java side? 

So the question is, are there any good reason, a fixed buffer is used instead? 
If not I would create a pull request for this, allowing variable sized data.

natdan, 2015-12-10:
I suspect that it was just least resistance path (at the point it was coded). Determining buffer size from Java side makes more sense. I think you should create pull request anyway and if Robert is OK with that we'll pull it in.

savageautomate, 2015-12-10:
Hi empirephoenix,

This may be a duplicate of issue: https://github.com/Pi4J/pi4j/issues/143
This has been fixed in the 'develop' branch and released in the 1.1-SNAPSHOT builds.  Please try the newer build to see if that fixes the problem for you.

Thank You,
Robert

empirephoenix, 2015-12-10:
Yes, you are right, it is already done in the develop. Thanks :+1:

closed 2015-12-10 by empirephoenix