Pi4J/pi4j-v1 #198: Problems with data from HMC5883L class

nkolban, 2015-12-14:
I am using the HMC5883L class to read from a HMC5883L device.  To achieve that task, I am calling readGyro() followed by access to the raw values from X, Y and Z.  What I am finding is that the X value is correct but Y and Z are incorrect.  It almost appears that the data being read from I2C has had two bytes skipped.   Here is a log from my HMC5883L device from an app written in C and using WiringPi - 

x=-133, y=192, z=-477 - angle=304.710668
x=-130, y=188, z=-476 - angle=304.663446
x=-135, y=189, z=-475 - angle=305.537678
x=-133, y=190, z=-478 - angle=304.992020
x=-136, y=187, z=-474 - angle=306.027374
x=-132, y=187, z=-479 - angle=305.217593

Here is the log from my Pi4J application:

X=-132 ,Y=-480, Z=8378
X=-135 ,Y=-475, Z=9662
X=-131 ,Y=-477, Z=9149
X=-131 ,Y=-477, Z=9146
X=-132 ,Y=-476, Z=9403

See how the X values match ... but the Java Y seems to be Z and the Z value from Java is somewhere else.

Attached is my Java app for recreation ...

[sample.txt](https://github.com/Pi4J/pi4j/files/60830/sample.txt)

gjwo, 2016-07-14:
Was this ever resolved? I am having similar problems with MCU9250 over I2C.

JoshLikesBeer, 2016-09-28:
I found this also. The values are (x, z, y) not (x, y, z), and need to be translated into ints from 16-bit two's complement values.

@savageautomate , I found a fix for this by modifying com.pi4j.component.gyroscope.honeywell.HMC5883L to add this method...
```
int twosCompliment16(int i) {
   if (i > 0x8000) {
      return i - 0x10000;
   }
   return i;
}
```

...and by changing lines 199-201 in readGyro() to look like this.
```
int x = twosCompliment16(((data[0] & 0xff) << 8) + (data[1] & 0xff));
int z = twosCompliment16(((data[2] & 0xff) << 8) + (data[3] & 0xff));
int y = twosCompliment16(((data[4] & 0xff) << 8) + (data[5] & 0xff));
```

savageautomate, 2016-12-02:
I applied you changes :-)  Thanks!

commit c915000c (2016-12-02): Fixed #198; Problems with data from HMC5883L class
closed 2016-12-02 by savageautomate