Pi4J/pi4j-v1 #203: Java 8

natdan, 2016-01-02:
Is there any reason why pi4j should be Java 1.7 bound?

savageautomate, 2016-01-11:
Traditionally for a library, I try to maintain one major version older than the currently released JVM.  So many tools and containers take a while to update to the latest JVM and just did not want it to be an obstacle when using Pi4J.  

Maybe with Java 8 now shipping on the Pi its time to consider updating?

Thanks, Robert

natdan, 2016-01-11:
Completely agree. Only difference now is that Java 8 brings quite a lot of new stuff directly applicable to many aspects of pi4j (namely closure/lambdas) and we're already on update #65. 

This is quite deliberately open question here so everyone can jump in and agree or oppose move. I am not in hurry to just jump the train and leave wreckage behind (incompatible code breakage, etc).

But, if nobody objects - why not? Any comments?

savageautomate, 2016-07-28:
Updated min requirement to Java 8 in version 1.2-SNAPSHOT
Via commit: https://github.com/Pi4J/pi4j/commit/51c1e9ac71414573e62c4c42714dddce36dc228c

closed 2016-07-28 by savageautomate