Pi4J/pi4j-v1 #206: SystemInfo Memory leak?

ddebrunner, 2016-01-06:
Just running this I see a slow memory leak, started out as using around 4.5% memory (as seen with top) and rose to 5.7% after 40+ minutes.

```
        for (;;) {
            SystemInfo.getCpuTemperature();
            SystemInfo.getCpuVoltage();
        }
```

I was seeing a memory leak in a longer running application that polled the system info and isolated it to this.

ddebrunner, 2016-01-06:
Seeing that the implementation is using Process to execute a command, it may be that it's actually a JVM issue.

savageautomate, 2016-04-11:
Unable to determine memory leak in Pi4J code.
Added a process.destroy() just in case.
https://docs.oracle.com/javase/7/docs/api/java/lang/Process.html#destroy()

Modified impl to use try-with-resources to better ensure closure of streams/readers.

commit 3a9db411 (2016-04-11): fixed #206; implemented try-with-resources and process destroy for ExecUtil.execute().
closed 2016-04-11 by savageautomate