Pi4J/pi4j-v1 #207: Interrupt-based GPIO is dead slow

hvesalai, 2016-01-10:
I'm trying to use Pi4J com.pi4j.wiringpi.GpioInterrupt to read GPIO pin for events at 2kHz (clock ticks), but unfortunately I'm only getting callbacks at the rate of 40Hz, i.e. very slowly. 

Reading the code, my quick guess is that it's the AttachCurrentThread/DetachCurrentThread pair in pi4j-native/src/main/native/com_pi4j_wiringpi_GpioInterrupt.c that is causing the slowness. It also causes each callback call to look like it's coming from a different thread (new name and id).

* Is there a reason why the thread is Detached after each callback?
* Is there a reason the poll loop is not written in Java with just the call to poll being native code. This way the Thread would be a normal Java thread, no need for Attach/Detach and the whole native code would be a one-liner?

hvesalai, 2016-01-11:
I got my system working fast enough for my use by modifying GpioInterrupt so that the Thread runs on Java side and native code is used only for polling. I.e. no calling java from native. I can push the modified code if anybody is interested.

natdan, 2016-01-11:
Cool. Of course we are all interested. Could you make a pull request, please, Robert can check the code before merging it?

savageautomate, 2016-01-11:
Hi hvesalai,

40Hz seems terribly slow.  I recently performed some testing and optimization on the latest 1.1-SNAPSHOT builds and achieved much higher results.  Please see the "GPIO Performance" section of this article: http://www.savagehomeautomation.com/projects/raspberry-pi-with-java-programming-the-internet-of-things-io.html

*Is there a reason why the thread is Detached after each callback?*

I don't remember with 100% certainty, but I think there was some problem when attempting to maintain the attached Java thread.  It may have been related to running under a Tomcat web server.  I think there was no guarantee that the Java thread was valid when then native code needed to perform the callback.     

*Is there a reason the poll loop is not written in Java with just the call to poll being native code. This way the Thread would be a normal Java thread, no need for Attach/Detach and the whole native code would be a one-liner?*

I think we originally wanted to tie into a non-polled implementation for GPIO change events something based on real ISRs and wanted to provide an abstraction point to easily adapt it later.  Polling was not the ideal solution but was all that was accessible at the time.

I would be interested in seeing your work and how you implemented it.  If you create a pull-request, please do so based on the 'develop' branch - there is where current changes are committed for the next release.

Thanks, Robert

hvesalai, 2016-01-11:
Sorry, this is now based on master

hvesalai, 2016-01-11:
I'll see about rebasing it to develop

hvesalai, 2016-01-11:
There, made a new pull request to the develop branch

hvesalai, 2016-01-12:
Btw, while I tested the callbacks, I personally use the polling interface I created directly myself. It simplifies my code because I need to wait for a start/stop bit (clock is high for > 2ms) and using poll with a timeout of 2ms is a great way to do this. With callbacks I would have to keep track of how long it was since the previous event.

natdan, 2016-01-12:
Is there possibility for both, polling and non polling implementations to exist next to each other? I am asking from perspective of multicore CPU where polling being done in one of four cores make little difference while polling in one core CPU can leave a dent in CPU usage. Also, there are other drawbacks with it namely different configurations for different implementations...

hvesalai, 2016-01-12:
@natdan I didn't change the implementation. By polling in this context we (I and the original implementation) mean `int poll(struct pollfd *fds, nfds_t nfds, int timeout);`. So there should be no new drawbacks, only positive things this change brings to you.

hvesalai, 2016-01-12:
I.e. there is still the

`public static synchronized void addListener(GpioInterruptListener listener)` 

besides the new

`public static native int pollPinStateChange(int pin, int previousValue, int pollTimeout)`

The CPU use is the same and is relative to the amount of events generated by the pin. No CPU is wasted if there are no events.

nzhome, 2016-07-24:
ouch, 40mhz is bad. thats 25milliseconds.    all GPIO should achieve less than 5ms latency, i.e. 5 times faster than that. is there anything I can do to help?  I might help write some sample code that tests the throughput of what you guys are building here.  Also this was Jan2016, has things improved since?

rlsutton1, 2016-07-24:
You mean 40hz not mhz.

salk31, 2017-11-05:
I've hit this one too. Were there any objections to the pull request?

savageautomate, 2021-01-12:
**closed**; all new development effort is moving to Pi4J v2.x.

closed 2021-01-12 by savageautomate