Pi4J/pi4j-v1 #213: How to read temperature, humidity from DHT22, using pi4j?

anhdenday, 2016-01-22:
Dear All, 

I would like to read data from sensor with Pi4J, my sensor using is DHT22, I've searched on the internet but very very few results with pi4j. Below link is an example but it's seem that Pi4j cannot do that? https://www.raspberrypi.org/forums/viewtopic.php?f=81&t=91896
I am not sure but I really with to do it with Pi4J because my project using java, so that I want to integrate Pi4J to read directly data from the devices!!

Please help me. Thanks you

gam2046, 2016-01-22:
I have tried to do so (DHT11), but without success. Because the response speed of the sensor is completed within a few microseconds. Thus, Java is not obtained the response information of the sensor, it has completed the acquisition of information.

If you find a way to use Java to get the temperature, I hope you can tell me.

natdan, 2016-01-22:
I think the way forward (at least in short term) is to do implementation similar to the one in the link (in c) and make JNI adapter to it. Are any of you guys up to do it or want me to do it and you to test it? 
I don't have DHT11/22 in my possession but I don't mind spending an hour or so for pi4j to get support for it.

gam2046, 2016-01-23:
I am willing to help you test. I have tested it myself, the same code with the Java can not get the sensor response signal, but Python can. So, I guess I/O is unable to keep up with a few microsecond intervals.

anhdenday, 2016-01-23:
@natdan: Thanks you, I am waiting for you!

anhdenday, 2016-01-29:
Dear All,
I found an another solution but I tested not successful, please check it for me. This [https://github.com/PoJD/hawa/blob/master/rpi/src/main/java/cz/pojd/rpi/sensors/gpio/Dht22Am2302TemperatureAndHumiditySensor.java](url) is also using Pi4J. I just make a main method to test it but it appear an error, please debug with me to find out what problem.
`pi@raspberrypi ~/pi4j $ java -cp rpi-1.0-SNAPSHOT.jar cz.pojd.rpi.sensors.gpio.Dht22Am2302TemperatureAndHumiditySensor
Exception in thread "main" java.lang.NoClassDefFoundError: com/pi4j/util/NativeLibraryLoader
	at cz.pojd.rpi.sensors.gpio.Dht22Am2302TemperatureAndHumiditySensor.<clinit>(Dht22Am2302TemperatureAndHumiditySensor.java:32)
Caused by: java.lang.ClassNotFoundException: com.pi4j.util.NativeLibraryLoader
	at java.net.URLClassLoader$1.run(URLClassLoader.java:366)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:355)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:354)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:425)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:308)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:358)`

While waiting for @natdan till he finish his code, please help me to check this for me!
Thanks

natdan, 2016-01-29:
While waiting on me I checked implementation over there that is pretty much what I was having in mind. I'll get in touch with Lubos and ask nicely if we is willing to contribute that impl to rpi.
In the mean time you need to include more jars when running it. I think following two would be enough:
pi4j-core-1.0.jar and maybe pi4j-gpio-extension-1.0.jar:
java -cp rpi-1.0-SNAPSHOT.jar:pi4j-core-1.0.jar:pi4j-gpio-extension-1.0.jar
(don't forget to dig them up from somewhere - [here](http://search.maven.org/#search|ga|1|g%3A%22com.pi4j%22)'s one place) and put them on your RPi

anhdenday, 2016-01-29:
Dear @natdan 
1- Actually I install maven in my raspberry pi 2 also to build directly on my device :D. So that i run "mvn install" first so the library should be download enough on the device, anyway I will try with your way. 
2- That's great if you and Lubos @PoJD  could contribute this implementation to p4j. 
Thanks you so much

anhdenday, 2016-02-06:
Hi Natdan,

anhdenday, 2016-02-06:
Hi Natdan,
How about your implementation and the contact status between you and Lubos?
Lasttime i just tried with rpi following your guide but it seem that rpi could not read the "libdht.so"! May be that file has problem and I dont know how to make it again on raspberry pi2. But i am have on a trip. I will comeback to work on this and report you soon!

Moshe-Malka, 2016-03-13:
does anybody have a solution to this problam ?

PoJD, 2016-03-13:
Hi,

Not sure what issue are you referring to, but I replied to all open tickets
in my github repo: https://github.com/PoJD/hawa/issues/79 and
https://github.com/PoJD/hawa/issues/78

Let me know if you need any further help.

Thanks!

anhdenday, 2016-03-15:
Dear @PoJD,
I still meet problem below when I try to run your code.

```
pi@raspberrypi:~/sensor/pi4j $ java -cp rpi-1.0-SNAPSHOT.jar:pi4j-core.jar:pi4j-gpio-extension.jar cz.pojd.rpi.sensors.gpio.Dht22Am2302TemperatureAndHumiditySensor
Unable to load the shared library dhtFile /home/pi/sensor/pi4j/libdht.so was not found in classpath.
Init failed before, not attempting to read anything from the sensor.
```
The file libdht.so is exactly in this path: " /home/pi/sensor/pi4j/libdht.so"

```
 static {
	try {
	  String path = "/home/pi/sensor/pi4j/libdht.so";
	    NativeLibraryLoader.loadLibraryFromClasspath(path);
	} catch (UnsatisfiedLinkError | IOException e) {
		System.out.println("Unable to load the shared library dht"+ e.getMessage());
	    initiated = false;
	}
    }
```

Thanks

PoJD, 2016-03-15:
Same issue again, you are changing pi4j code path and loading it from that
ppath, but the command you launch java does not include that path in
classpath.

Either add the path to that dir to the cp  switch on command line or  do
not change pi4j code and use the default path the code expects (and still
add that path to cp  switch).

anhdenday, 2016-03-15:
Sorry, I did not get your point, I did not change anything in Pi4j code. 
I just make a main method at the file Dht22Am2302TemperatureAndHumiditySensor.java to test with the command "java -cp Dht22Am2302TemperatureAndHumiditySensor FilePath.Java"
In my command already include pi4j: 
"java -cp rpi-1.0-SNAPSHOT.jar:pi4j-core.jar:pi4j-gpio-extension.jar"
Thanks

PoJD, 2016-03-15:
OK, sorry, I missed that it is your code that attempts loading the lib
directly.

My point re classpath still stands though. The path /home/.../sensor is not
in your classpath, do jvm cannot load the library.

I would need to check at my PC though how is this supposed to work. I do
not think such loading logic is needed in client code, rpi project should
do it by its own I think.

anhdenday, 2016-03-15:
I try to change "NativeLibraryLoader.loadLibraryFromClasspath(path);" by " System.load(path);" 
I think it's load SUCCESS already but it appear new problem!
Below is my code!

   ```
 static {
	try {
	    String path = "/home/pi/sensor/pi4j/libdht.so";
	    System.load(path);
	    System.out.println("LOAD SUCCESS");
	    
	} catch (Exception e) {
		System.out.println("Unable to load the shared library dht"+ e.getMessage());
	    initiated = false;
	}
    }
```

 ```
@Override
    public List<Reading> readAll() {
	List<Reading> result = new ArrayList<>();
	if (isInitiated()) {
	    double[] numbers = executeNative(gpioPin);
	    System.out.println("Numbers is: " + numbers);
	    
	    if (numbers != null && numbers.length == 2) {
		result.add(Reading.newBuilder().type(Type.humidity).doubleValue(numbers[1]).units("%").build());
	    } else {
	    System.out.println("Some error ocurred when attempting to read the values from the sensor. Output: " + Arrays.asList(numbers));
	    }
	} else {
		System.out.println("Init failed before, not attempting to read anything from the sensor.");
	}
	return result;
    }
```
```
 public static void main(String[] args) {
		Dht22Am2302TemperatureAndHumiditySensor t = new Dht22Am2302TemperatureAndHumiditySensor(4);
		for (Reading r: t.readAll()){
			System.out.println("Reading data: " + r.toString());
		}
	}
```

After running this code it appear below error.

```
pi@raspberrypi:~/sensor/pi4j $ java -cp rpi-1.0-SNAPSHOT.jar:pi4j-core.jar:pi4j-gpio-extension.jar cz.pojd.rpi.sensors.gpio.Dht22Am2302TemperatureAndHumiditySensor
LOAD SUCCESS
Numbers is: null
Some error ocurred when attempting to read the values from the sensor. Output: [null]
```

Thanks you so much

yglodt, 2016-04-25:
I am planning a project in Java where I need to get data from 4 DHT22. Is there a chance I can use pi4j or should I exec a Python script based on the Adafruit library?

nyaria, 2016-04-29:
I've spent lot of time to make a working solution in Java for DHT22.
Pi4J failed because the timing wasn't precise enough to accurately read the sensor data.
After that I made a JNI wrapper for the C code. It worked sometimes and sometimes not (mostly). The same C code worked perfectly without the JNI wrapper.
Thus I made a wrapper in Java for the python code.
A few month later I found a new sensor AM2321. Allegedly it is I2C compatible but it quickly turned out it doesn't. Luckily the C code and the JNI wrapper worked perfectly in this case.
So I still use this configuration.

yglodt, 2016-04-29:
Thanks @nyaria for your answer.

I was under the (maybe wrong) impression that pi4j uses the wirpingpi library via JNI, and thus the timing issue should not happen because all is done in native C code.

I can still use the Adafruit Python lib and exec() it from java, or even the the wiringpi rht binary...

anhdenday, 2016-04-30:
Hi @nyaria ,
Could you please share your solution with JNI Wrapper and C/Python code?
Thanks

L3vi4th4n, 2016-08-07:
Hello,

I'm also trying to read values from a DHT11.

I found a snippet on this [stackoverflow thread](http://stackoverflow.com/questions/28486159/read-temperature-from-dht11-using-pi4j/34976602#34976602) that I modified a bit, but I do not manage to make it work.
I'm using the last version of the pi4J library (1.1) and the last version of the JVM provided on Raspbian.

The problem always comes from the WiringPi interface, but I do not manage to make it work.
If anyone sees a problem in this code, I would be glad to know it.

I'm running my program as root for information :)

Modified code :
```
import org.apache.logging.log4j.LogManager;
import org.apache.logging.log4j.Logger;

import com.pi4j.io.gpio.Pin;
import com.pi4j.wiringpi.Gpio;
import com.pi4j.wiringpi.GpioUtil;

public class DHT11 {
	static final Logger LOG = LogManager.getLogger(DHT11.class);
	private static final int MAXTIMINGS = 85;
	private int[] dht11_dat = { 0, 0, 0, 0, 0 };
	private Pin sensorPin;

	public DHT11(Pin gpioPin) {

		this.sensorPin = gpioPin;

		/* The following code should be done but generates an error... */
		// // setup wiringPi
		// if (Gpio.wiringPiSetup() == -1) {
		// System.out.println(" ==>> GPIO SETUP FAILED");
		// return;
		// }

		try {
			// Generates a fatal error in JNI --> SIGSEGV --> C
			// [libpi4j4846801952720539391.so+0xbf3c] wpiPinToGpio+0x10
			GpioUtil.export(gpioPin.getAddress(), GpioUtil.DIRECTION_OUT);
		} catch (Exception e) {
			LOG.error(e);
		}
	}

	public DHT11Data getSensorReadings() {
		DHT11Data result = null;
		int laststate = Gpio.HIGH;
		int j = 0;
		int counter = 0;
		dht11_dat[0] = dht11_dat[1] = dht11_dat[2] = dht11_dat[3] = dht11_dat[4] = 0;

		// pull pin down for 18 milliseconds
		Gpio.pinMode(sensorPin.getAddress(), Gpio.OUTPUT);
		Gpio.digitalWrite(sensorPin.getAddress(), Gpio.LOW);
		Gpio.delay(18);

		// then pull it up for 40 microseconds
		Gpio.digitalWrite(sensorPin.getAddress(), Gpio.HIGH);
		Gpio.delay(40);

		// prepare to read the pin
		Gpio.pinMode(sensorPin.getAddress(), Gpio.INPUT);
		// detect change and read data
		for (int i = 0; i < MAXTIMINGS; i++) {
			counter = 0;
			while (Gpio.digitalRead(sensorPin.getAddress()) == laststate) {
				counter++;
				Gpio.delayMicroseconds(1);
				if (counter == 255) {
					break;
				}
			}

			laststate = Gpio.digitalRead(sensorPin.getAddress());

			if (counter == 255) {
				break;
			}

			// ignore first 3 transitions
			if (i >= 4 && i % 2 == 0) {
				// shove each bit into the storage bytes
				dht11_dat[j / 8] <<= 1;
				if (counter > 16) {
					dht11_dat[j / 8] |= 1;
				}
				j++;
			}
		}
		// check we read 40 bits (8bit x 5 ) + verify checksum in the last byte
		// print it out if data is good
		if (j >= 40 && checkParity()) {
			float tempDegCel = dht11_dat[2] + 0.1f * dht11_dat[3];
			float humidity = dht11_dat[0] + 0.1f * dht11_dat[1];

			result = new DHT11Data(tempDegCel, humidity);

		} else {
			LOG.debug("Data not good, skip");
		}

		return result;

	}

	private boolean checkParity() {
		return dht11_dat[4] == (dht11_dat[0] + dht11_dat[1] + dht11_dat[2] + dht11_dat[3] & 0xFF);
	}

	public class DHT11Data {

		/** Temperature in °C */
		private final float temperature;
		/** Humidity */
		private final float humidity;

		/**
		 * 
		 * @param tempDegCel
		 *            The temperature in °C
		 * @param humidity
		 *            The humidity value in percent
		 */
		public DHT11Data(float tempDegCel, float humidity) {
			this.temperature = tempDegCel;
			this.humidity = humidity;
		}

		/**
		 * Method to get the temperature in °C
		 * 
		 * @return The temperature value in °C
		 */
		public float getTemperatureCelsius() {
			return temperature;
		}

		/**
		 * Method to get the temperature in °F
		 * 
		 * @return The temperature value in °F
		 */
		public float getTemperatureFahrenheit() {
			return temperature * 1.8f + 32;
		}

		/**
		 * Method to get the humidity.
		 * 
		 * @return The humidity value in percent
		 */
		public double getHumidity() {
			return humidity;
		}

	}
}
```

rlsutton1, 2016-08-08:
Java us not a realtime language and can not do the timing accuracy required.

L3vi4th4n, 2016-08-08:
I agree. But is it not compensated by using the wiringPi encapsulated in pi4j ?
The matter here is that I cannot even manage to initialize wiringPi (executed in root, pls see comments in code)

Thx

yglodt, 2017-04-08:
FYI the code (DHT11.java) in the post below is producing correct readings half of the time on a Pi3. not so bad at the end!

http://stackoverflow.com/a/34976602/272180

savageautomate, 2021-01-15:
**Closed.**  No further activity.

closed 2021-01-15 by savageautomate