Pi4J/pi4j-v1 #214: Gpio.wiringPiISR() causing crash of Java Runtime Environment

HapibotDinis, 2016-01-22:
I'm using 1.1-SNAPSHOT build and the application crashes when receiving an interruption defined as the one below:

Gpio.wiringPiISR(MIC_PIN, Gpio.INT_EDGE_FALLING,new GpioInterruptCallback() {
    ... 
});

This error has been reported in previous issues (#180, #176, #182) and the solution was to use the latest snapshot, which I did.
I've attached an error log.
Any solution will be appreciated. Thank you.
[hs_err_pid2976.txt](https://github.com/Pi4J/pi4j/files/101022/hs_err_pid2976.txt)

savageautomate, 2016-03-18:
I am actively looking into this issue.  I did not experience a crash, but its was not working properly either.  Looking deeper into the issue tomorrow.  Thx!

savageautomate, 2016-03-18:
I have attempted to fix this issue.  It seem to be working for me now.  Please try and reopen the issue if you still see a problem.   Thanks, Robert

commit 098baccf (2016-03-18): Fix #214; reworked the logic for the ISR callbacks; simplified by moving user callback references to Java and allowing the native code to call a fixed static callback method in the Gpio class rather than attempting to directly find and invoke the user's callback reference via JNI.
closed 2016-03-18 by savageautomate