Pi4J/pi4j-v1 #217: RaspiPin.getPinByName("GPIO 2") returns null in 1.1-SNAPSHOT

mstevetodd, 2016-02-02:
Calling RaspiPin.getPinByName("GPIO 2") (or any other pin by name) in 1.1-SNAPSHOT returns null, _unless_ one of the static values has been accessed earlier.  
For example, in ListenGpioExample.java, changing the line 
`final GpioPinDigitalInput myButton = gpio.provisionDigitalInputPin(RaspiPin.GPIO_02, PinPullResistance.PULL_DOWN);`
to
`final GpioPinDigitalInput myButton = gpio.provisionDigitalInputPin(RaspiPin.getPinByName("GPIO 2"), PinPullResistance.PULL_DOWN);`
results in an error due to RaspiPin.getPinByName("GPIO 2") returning null.
If I add a line like:
`Pin tp = RaspiPin.GPIO_02;`
prior to the line above, the getPinByName works properly.  It seems that something isn't being initialized, but I'm not sure what that is.
This syntax works fine in 1.0.  
I've also compiled the latest development branch but it exhibits the same problem.

Regards,
  SteveT

sanados, 2016-03-14:
This issue does not exist in master as RaspiPin inits the pins static on declaration.

Just commented on issue #224 about this.
The bug was fixed, but it seems that snapshot 1.1 was built from a very old version.

sanados, 2016-03-14:
One thought though.

In RaspiPin line 47:

`private static Map<String, Pin> pins = new HashMap<String, Pin>();`

I would replace with

`private static final Map<String, Pin> pins = new HashMap<>();`

savageautomate, 2016-03-15:
So the issue is that all pins defined in the RaspiPin class are defined using static final members.  However, when your code is fist calling `RaspiPin.getPinByName`, the **getPinByName** is a static method actually defined in a parent class and the static **RaspiPin** members are not getting initialized.  Static methods in a parent are not technically inherited by the child class.  So in this case the child class **RaspiPin** is never initializing its static members.  

When you include your `Pin tp = RaspiPin.GPIO_02;` line before attempting to call `RaspiPin.getPinByName` that code will cause the static members of **RaspiPin** to get initialized.  

To make this easier for users to consume and not have to worry about the static initialization life-cycles, I'll add pseudo overriding methods to the Pin classes such as RaspiPin to ensure the actual **Pin class gets initialized.

closed 2016-03-15 by savageautomate

savageautomate, 2016-03-15:
new 1.1-SNAPSHOT build posted with these changes.

mstevetodd, 2016-03-15:
Thank you for the fix and especially the explanation of why the previous code wasn't working.

commit 75410807 (2016-03-15): Fixed #217 : Added static pseudo-overriding methods to ***Pin classes to allow for static initialization of the class members.  Prior to this change, calling static method belonging to the parent class (PinProvider) would not invoke any of the static members of the ***Pin class instances.