Pi4J/pi4j-v1 #222: Unnecessary GpioPinExistsException

Benjoyo, 2016-03-12:
I'm referring to the **GpioControllerImpl** and the method **provisionPin(GpioProvider provider, Pin pin, String name, PinMode mode, PinState defaultState)**.
Currently a **GpioPinExistsException** is thrown if the same pin is provisioned more than once.
Wouldn't it make more sense to throw an exception only if a pin is tried to be provisioned with a different **PinMode** than it was previously provisioned with and simply return the existing **GpioPin** instance? This would eliminate the need for a user to store provisioned pins in e.g. a singleton if they are needed at different parts of the program.

sanados, 2016-03-14:
But with your suggestion you could change mode and state of a pin that is provisioned already and that may cause you more problems than this rework would benefit.

savageautomate, 2016-03-15:
The verb/term "_**provision**_" is an explicit and intentional action that should only be performed if a given pin had not already been provisioned.  In order to re-provision a pin, the pin must first explicitly be un-provisioned. 

Consequently, I did go ahead and add support for getting an existing pin instance from the already provisioned pins in the **GpioController** to help address your use case.

See: https://github.com/Pi4J/pi4j/commit/b5cc65876a87c21ea9667900aeba39d4c2b8f1ac#diff-b267c565952860c625d706be9b87a425L74

Thanks, Robert

closed 2016-03-15 by savageautomate

savageautomate, 2016-03-15:
new 1.1-SNAPASHOT build posted with these changes.

commit b5cc6587 (2016-03-15): Fixed #222 : Added support for getting a provisioned pin by Pin instance or by name from the GpioController singleton.