Pi4J/pi4j-v1 #224: ArrayIndexOutOfBoundsException instantiating PiFaceDevice

CrashCash, 2016-03-14:
This may be related to "MCP23008GpioProvider. - java.lang.ArrayIndexOutOfBoundsException: 128" as it appears to be a pin caching issue.

     1	package pifacetest;
     2
     3	import com.pi4j.component.light.LED;
     4	import com.pi4j.component.relay.Relay;
     5	import com.pi4j.device.piface.impl.PiFaceDevice;
     6	import com.pi4j.io.gpio.GpioController;
     7	import com.pi4j.io.spi.SpiChannel;
     8	import java.io.IOException;
     9
    10	public class PiFaceTest {
    11
    12	    static int SECONDS = 1000;
    13
    14	    public static void main(String[] args) throws IOException, InterruptedException {
    15	        PiFaceDevice piface = new PiFaceDevice(SpiChannel.CS0);
    16
    17	        Relay relay = piface.getRelay(0);
    18	        LED led = piface.getLed(0);
    19
    20	        relay.close();
    21	        led.on();
    22
    23	        Thread.sleep(1 * SECONDS);
    24
    25	        relay.open();
    26	        led.off();
    27
    28	        GpioController gpio = piface.getGpio();
    29	        gpio.shutdown();
    30	    }
    31	}

Exception in thread "main"
java.lang.ArrayIndexOutOfBoundsException: 1001
at com.pi4j.io.gpio.GpioProviderBase.getPinCache(GpioProviderBase.java:69)
at com.pi4j.io.gpio.GpioProviderBase.export(GpioProviderBase.java:98)
at com.pi4j.gpio.extension.piface.PiFaceGpioProvider.export(PiFaceGpioProvider.java:247)
at com.pi4j.io.gpio.GpioProviderBase.export(GpioProviderBase.java:79)
at com.pi4j.io.gpio.impl.GpioPinImpl.export(GpioPinImpl.java:158)
at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionPin(GpioControllerImpl.java:533)
at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionPin(GpioControllerImpl.java:506)
at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionPin(GpioControllerImpl.java:501)
at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionDigitalInputPin(GpioControllerImpl.java:613)
at com.pi4j.device.piface.impl.PiFaceBase.<init>(PiFaceBase.java:74)
at com.pi4j.device.piface.impl.PiFaceDevice.<init>(PiFaceDevice.java:51)
at pifacetest.PiFaceTest.main(PiFaceTest.java:15)

This is using the 1.1 snapshot. This runs in 1.0 but does not exit.

sanados, 2016-03-14:
Already fixed in master it seems (reworked Array to ConcurrentHashMap) by Savage on 28.12.2014

but I can see 

 protected GpioProviderPinCache[] cache = new GpioProviderPinCache[100];
in the snapshot which is weird

CrashCash, 2016-03-14:
On 03/14/2016 06:36 AM, sanados wrote:
> Already fixed in master it seems (reworked Array to ConcurrentHashMap) by
> Savage on 28.12.2014
>
> but I can see
>
> protected GpioProviderPinCache[] cache = new GpioProviderPinCache[100];
> in the snapshot which is weird

Soooo.... is it still a bug, or is it resolved? And if it's resolved, where do 
I get a snapshot?

-- 
RMS-S-OK_RRL, record locked against read but read anyway
-- real VMS error message

savageautomate, 2016-03-15:
So the issue here is that in an optimization effort for pin state switching performance we moved back from the ConcurrentHashMap implementation of the pin cache storage to an array in 1.1-SNAPSHOT.  

However, the array has a maximum defined size of 100 elements and pins are referenced by their pin address.  There are no devices (supported) that currently use over 100 pins, so all should be good right?  Well .. no ... in the case of the PiFace and some other GPIO expansion devices the pin address variable is assigned much higher non-contiguous values and thus the `java.lang.ArrayIndexOutOfBoundsException` occurs when the pin cache is accessed for these pin's with addresses above a value of 100.  

See example: https://github.com/Pi4J/pi4j/blob/develop/pi4j-gpio-extension/src/main/java/com/pi4j/gpio/extension/piface/PiFacePin.java#L60

I have modified the pin cache to implement a dynamic expanding array to accommodate these cases where the pin addresses may be higher than the default allocation capacity of 100.

closed 2016-03-15 by savageautomate

savageautomate, 2016-03-15:
new 1.1-SNAPASHOT build posted with these changes.

wouterpol93, 2016-06-01:
Facing the same issue in the PCA9685GpioProvider, are there any fixes for that?

rlsutton1, 2016-06-01:
Unless i am missing something... The removal of concurrenthashmap stops the pin cache being thread safe. When working with multiple threads on multiple cores you will sooner or later see stale values in the cache.

commit 354dea65 (2016-03-15): Fixed #224 : Modified the pin cache to implement a dynamic expanding array to accommodate cases where the pin addresses may be higher than the default allocation capacity of 100.