Pi4J/pi4j-v1 #225: REPEATED START

HarrySatt, 2016-03-14:
Hi there
I am looking for a way to programmatically ( with Pi4J ) switch on or off the use of REPEATED START. Some devices need to be addressed by a REPEATED START sequence otherwise they don not send out valid data. One of theses is the device used in the sample code from ORACLE. Xtrinsic MMA8452Q 3-Axis,12-bit/8-bit Digital Accelerometer. Breakoutboard build by SPARCDFUN open hardware.
I know how to switch REPEATED START on and off from bash but would like to know how it can be done from JAVA. Asking which mode is straight forward.

```
public static boolean IsI2C_Combined()
    {
        boolean ret = true;
        Reader in = null;
        CharBuffer cb = CharBuffer.allocate(100);
        
        Path path = FileSystems.getDefault().getPath("/sys/module/i2c_bcm2708/parameters", "combined");
        System.out.println(path.toString());
        try {
            in = Files.newBufferedReader(path);
            in.read(cb);
            if(cb.get(0) != 'Y')
            {
                ret = false;
            }
        }
        catch(IOException ex)
        {
            System.out.println("Error in reading COMBINED Value");
        }
        finally
        {
            if( in != null )
            {
                try {
                    in.close();                      
                }
                catch( IOException ex)
                {
                    System.err.println("What ist that?");
                }
            }
        }
        return ret;
    }
```

Or is there a better way also for this?

Thanx, Harald.

natdan, 2016-03-14:
It seems slightly more complicated than I would do it (but otherwise fine). Here is what I would do:

	public static boolean isI2CCombined() {
		try (FileInputStream in = new FileInputStream("/sys/module/i2c_bcm2708/parameters/combined")) {
		  return in.read() == 'Y';
		} catch (IOException ignore) {
			System.err.println("Cannot read combined");
		}
	}

or if you don't want Java 8:

	public static boolean isI2CCombined() {
		try {
			FileInputStream in = new FileInputStream("/sys/module/i2c_bcm2708/parameters/combined");
			try {
			  return in.read() == 'Y';
			} finally {
				in.close();
			}
		} catch (IOException ignore) {
			System.err.println("Cannot read combined");
		}
		return false;
	}

BTW by convention Java methods start with lower case and they don't have '_' (read [here](http://www.oracle.com/technetwork/java/codeconventions-150003.pdf))

HarrySatt, 2016-03-15:
Hi.
I am looking for a way to set or clear the combined parameter easily from within a JAVA program. The code snipet only shows where to get the info if REPEATED START is used or not.
Thinks for the answer.
Harald

natdan, 2016-03-15:
Just reverse it. Do you want me to paste code snipped that can do that? I don't mind. Also, it is good idea to include it in pi4j. I'll look into it, too!

HarrySatt, 2016-03-15:
Hi natdan. Thanks. That works. I was up to now a bit confused about setting this combined parameter.
Narmally I used < echo -n 1 >combined > from bash and tried to mimic this in JAVA, But I need to write out Y or N character.
Now it works. Would be a nice feature in Pi4J.
```
    public static boolean IsI2cCombined()
    {
        boolean ret = false;
        try (FileInputStream in = new FileInputStream("/sys/module/i2c_bcm2708/parameters/combined")) {
            ret = in.read() == 'Y';
        } catch (IOException ignore) {
            System.err.println("Cannot read combined");
        }
        return ret;
    }
```
```
public static boolean SetI2cCombined(boolean status)
{
        boolean ret = false;
        try (FileOutputStream out = new FileOutputStream("/sys/module/i2c_bcm2708/parameters/combined")) {
            out.write(status?'Y':'N');
            ret = true;
        }
        catch(IOException ignore) {
            System.err.println("Can not set combined");
        }
        return ret;
}
```

natdan, 2016-03-15:
Yes. That looks correct. And, yes, you are correct it should end in pi4j. I'll look to it. 

BTW - method names should really read isI2CCombined() and setI2CCombined().

Also, it is acceptable not to catch IOException but just propagate it out. Something like:

    public static void SetI2cCombined(boolean status) throws IOException {
        try (FileOutputStream out = new FileOutputStream("/sys/module/i2c_bcm2708/parameters/combined")) {
            out.write(status?'Y':'N');
        }
    }

    public static boolean IsI2cCombined() throws IOException {
        try (FileInputStream in = new FileInputStream("/sys/module/i2c_bcm2708/parameters/combined")) {
            return in.read() == 'Y';
        }
    }

And no need to return value. I think I'll just use this code and add it to appropriate place. Give me a day or two.

HarrySatt, 2016-03-15:
OK. Fine. I wil use this naming convention in the future. I am fresh in JAVA and programmed C++ up to now. There it was allowed to use underscore. Thanks a lot for your help and integrating this REPEATED START dingi into Pi4J.
Harald.

savageautomate, 2021-02-15:
No further activity; issue closed

closed 2021-02-15 by savageautomate