Pi4J/pi4j-v1 #226: Add support for Raspberry Pi 3B

savageautomate, 2016-03-16:
## Issue

Officially add support for Raspberry Pi 3B

## Progress
The following steps need to be completed to officially announce support for the 3B.

- [X] update wiringPi to latest version
- [X] test/validate GPIO outputs
- [X] test/validate GPIO inputs with pull down
- [X] test/validate GPIO inputs with pull up
- [X] test/validate serial communication -- **KNOWN ISSUE, SEE BELOW**
- [X] test I2C communication
- [ ] test SPI communication
- [X] update project readme
- [ ] update project documentation

## Summary

With the latest builds of 1.1-SNAPSHOT, Pi4J is highly functional with the RaspberryPi 3B.  A bit more testing is needed as well as adding documentation.  The pin numbering scheme for the 3B is exactly the same as the 2B and listed here: http://pi4j.com/pins/model-2b-rev1.html

## Known Issues

Serial/UART device /dev/ttyAMA0 on Raspberry Pi 3B is not working properly due to a clock timing issue.
A workaround is to configure force_turbo=1 in /boot/config.txt
(see: https://github.com/RPi-Distro/repo/issues/22)

emeflag, 2016-03-25:
Robert... I'm very interested in trying out Pi4J on a raspberry pi 3.  Is it possible to get the latest snapshot to give it a test drive.  If so where might it be located. Thanks so much for creating a wonderful java interface to GPIO. Cheers... Eric

savageautomate, 2016-03-28:
@emeflag Yes!  You can download the latest SNAPSHOT builds from this page:
http://pi4j.com/download.html

Or if you are using Maven with your project, you can reference the dependencies like this:
```
<dependency>
    <groupId>com.pi4j</groupId>
    <artifactId>pi4j-core</artifactId>
   <version>1.1-SNAPSHOT</version>
</dependency>
```

And add the SNAPSHOT repository to your Maven POM:

```
<repositories>
	<repository>
		<id>oss-snapshots-repo</id>
		<name>Sonatype OSS Maven Repository</name>
		<url>https://oss.sonatype.org/content/groups/public</url>
		<snapshots>
			<enabled>true</enabled>
			<updatePolicy>always</updatePolicy>
		</snapshots>
	</repository>
</repositories>
```

emeflag, 2016-03-28:
Thanks so much.  Maven sounds like something I might want to learn how to use.

savageautomate, 2016-03-28:
If you are going to spend the time learning a build environment/ecosystem, I might suggest Gradle over Maven these days.  There is a way to reference dependencies from a Maven repository in a Gradle build as well.  

Thanks, Robert

emeflag, 2016-03-28:
Robert….

Looking forward to giving your Pi4J package a try.  We're exploring the possibility for a robotic-team summer project of launching a weather balloon carrying a raspberry pi hosting a complement of sensors, camera, and GPS. Pi4J may be the ticket as we have java student programmers. 

I am a programming mentor for two robotics teams. 

We have a great group of student programmers on two robotics teams:  FIRST FRC Coconuts team 2486 (http://www.fusd1robotics.org)  and FIRST FTC Astronuts Team 3486, both at Coconino High School, Flagstaff AZ.

The FRC Coconuts program in java on a National Instruments “RoboRio" processor using the FRC java programming environment. They use eclipse as their IDE.  http://wpilib.screenstepslive.com/s/4485/m/13809

The FTC Astronuts program their robot in java using Android phones as the control system. They use Android Studio as the IDE.
http://www.first-tech-challenge.org/en/

Cheers…  Eric

codehub, 2016-05-08:
Hey Robert, I'm new to Pi and pi4j. Just started playing around with my pi 3 and sample projects and wanted to thank you so much for your efforts on this. Regards, Sam

Ettery, 2016-05-20:
Second that, thank you for all your efforts on the Robert!

savageautomate, 2016-05-20:
@Ettery & @codehub 

Thanks!  Hopefully I can get this version officially released soon.  Been very busy on other projects lately :-)

as-qs, 2016-06-09:
I just want to thank everyone who's been working on this awesome library. I've been using it for the past 2 years, and it's been running great. 

One question though, I've been experiencing some issues with SPI on the 4.4 kernel since it no longer supports the spi-bcm2708 driver. Is that something that would be resolved with the latest pi4j? Anyone else experiencing similar problems?

savageautomate, 2016-06-13:
I'm not sure of the issues you are getting with latest SPI driver.  It's unlikely (_although not impossible_) to be a Pi4J issue since we simply wrap the native API for accessing and communicating with the Linux SPIDEV driver.

savageautomate, 2016-06-13:
Updated site docs for Raspberry Pi 3B (v1.1) release.

closed 2016-06-13 by savageautomate

freemoo, 2017-04-17:
I'm trying work with the RC522 reader , is there any source example that I can work on it ?