Pi4J/pi4j-v1 #227: Add support for Raspberry Pi Zero

savageautomate, 2016-03-16:
## Issue

Officially add support for Raspberry Pi Zero

## Progress
The following steps need to be completed to officially announce support for the Pi Zero.

- [X] update wiringPi to latest version
- [X] test/validate GPIO outputs
- [X] test/validate GPIO inputs with pull down
- [X] test/validate GPIO inputs with pull up
- [ ] test/validate serial communication
- [ ] test/validate  I2C communication
- [ ] test/validate  SPI communication
- [X] update project readme
- [ ] update project documentation

## Summary

With the latest builds of 1.1-SNAPSHOT, Pi4J is highly functional with the RaspberryPi Zero.  A bit more testing is needed as well as adding documentation.  The pin numbering scheme for the Pi Zero is exactly the same as the 2B and listed here: http://pi4j.com/pins/model-2b-rev1.html

## Known Issues

None.

truell20, 2016-05-29:
What is the update on this? I would like to use Pi4J on the Rasberry Pi Zero. Is implemented and tested?

savageautomate, 2016-06-03:
Yes, it is implemented and tested on the latest 1.1-SNPSHOT builds.  
Documentation still needs to be updated.

savageautomate, 2016-06-13:
Updated site docs for Raspberry Pi Zero (v1.1) release.

closed 2016-06-13 by savageautomate