Pi4J/pi4j-v1 #229: Add support for Odroid C1+, C2, XU4

savageautomate, 2016-03-16:
## Issue

Officially add support for Odroid C1+, C2 and XU4 boards.

## Progress
The following steps need to be completed to officially announce support for the Odroid boards.

- [X] source Odroid C1+ hardware
- [ ] source Odroid C2 hardware
- [X] source Odroid XU4 hardware
- [X] include wiringPi for Odroid in build
- [X] package wiringPi for Odroid compiled SO libs in JAR
- [ ] test/validate GPIO outputs on Odroid XU4

- [X] test/validate GPIO outputs on Odroid C1+
- [X] test/validate GPIO inputs with pull down on Odroid C1+
- [X] test/validate GPIO inputs with pull up on Odroid C1+
- [ ] test/validate serial communication on Odroid C1+
- [ ] test/validate  I2C communication on Odroid C1+
- [ ] test/validate  SPI communication on Odroid C1+
- [ ] add project examples for Odroid C1+  **some examples have been created**

- [ ] test/validate GPIO outputs on Odroid C2
- [ ] test/validate GPIO inputs with pull down on Odroid C2
- [ ] test/validate GPIO inputs with pull up on Odroid C2
- [ ] test/validate serial communication on Odroid C2
- [ ] test/validate  I2C communication on Odroid C2
- [ ] test/validate  SPI communication on Odroid C2

- [ ] test/validate GPIO outputs on Odroid XU4
- [ ] test/validate GPIO inputs with pull down on Odroid XU4
- [ ] test/validate GPIO inputs with pull up on Odroid XU4
- [ ] test/validate serial communication on Odroid XU4
- [ ] test/validate  I2C communication on Odroid XU4
- [ ] test/validate  SPI communication on Odroid XU4

- [ ] update project readme
- [ ] update project documentation

## Summary

With the latest builds of 1.1-SNAPSHOT, Pi4J includes support for GPIO on the Odroid C1+.  This feature is experimental and any feedback is greatly appreciated.  

## Known Issues

Much

stefanhaustein, 2016-05-22:
Does this work in Odroid Android?

savageautomate, 2016-06-13:
@stefanhaustein 

No, this library has not been tested on an Android system.  It's a standard Java library that wraps a native C library using JNI.  I'm not sure the C library (WiringPi) is ported for Android?

Thanks, Robert

closed 2016-07-26 by savageautomate

savageautomate, 2017-08-06:
EDIT.

I have updated the WiringOP library to support the custom SPI modes.
via pull request: https://github.com/hardkernel/wiringPi/pull/5