Pi4J/pi4j-v1 #230: Add support for OrangePi.

savageautomate, 2016-03-16:
## Issue

Officially add support for OrangePi.

## Progress
The following steps need to be completed to officially announce support for the OrangePi boards.

- [X] source OrangePi hardware
- [ ] include wiringPi for OrangePi in build
- [ ] package wiringPi for OrangePi compiled SO libs in JAR
- [ ] test/validate GPIO outputs
- [ ] test/validate GPIO inputs with pull down
- [ ] test/validate GPIO inputs with pull up
- [ ] test/validate serial communication
- [ ] test/validate  I2C communication
- [ ] test/validate  SPI communication
- [ ] add project examples
- [ ] update project readme
- [ ] update project documentation

## Summary

I have the hardware for testing and have started investigating the viability of the WiringPi for Odroid library.  

## Known Issues

None

icebergx5, 2016-04-14:
Hi,
When Release 1.1 will be completed with OrangePi support?

Thanks.

DamienMiras, 2016-11-28:
I got an orange pi and some i2c devices;  I could test or implement that part :
- [ ] test/validate SPI communication
What I would need is a "getting started" doc for the wiring, C cross compilation tool chain, let me know how I could help

savageautomate, 2016-12-01:
Let me start working on getting the basic plumbing started and then you can implement/test/fix I2C/SPI related functionality.  Hopefully, I can get started next week.

savageautomate, 2016-12-06:
I have added experimental support for OrangePi in the latest 1.2-SNAPSHOT builds.  

Please be aware that this is largely untested but I was able to get all the build process worked out to include the WiringPi port for OrangePi.  

I'm afraid that the WiringPi port for OrangePi may be slightly outdated and may not support the newer/latest OrangePi hardware models, but more testing is needed to determine which models are working properly and which are not.

DamienMiras, 2016-12-06:
Thanks, just read the commits. all seems clear from now. I will pull this week-end to see if I can test. It may end up with maintaining orangePi wiringPi  impl... :)

savageautomate, 2016-12-10:
I have not done much testing yet, but I did find a couple of cursory errors and published these fixes in a new 1.2 snapshot build.  

Also, I had to switch to a distro that enables the GPIO sysfs capability.  I'm currently using Armbian.

OPI-0, 2017-03-11:
Hi!, 
I was wondering if PI4j will also work on the OPI Zero...If needed i can test with my OPI Hardware....
(need some help wit coding examples...)

Thanks!!

savageautomate, 2017-09-05:
OPI Zero is not supported in the current native builds because we compiled against:
https://github.com/zhaolei/WiringOP.git
and OPI Zero is not there.

However, see this issue for a workaround: https://github.com/Pi4J/pi4j/issues/333

savageautomate, 2021-01-12:
**CLOSED**; all new development efforts are in Pi4J v2.x (https://v2.pi4j.com)

closed 2021-01-12 by savageautomate