Pi4J/pi4j-v1 #234: Add support for SC16IS7xx - I2C/uart bridge

DrunkedBrigand, 2016-04-22:
Please add support for SC16IS7x I2C/uart bridge. I think many people wants more serial ports in their Raspberry Pi.

colinrgodsey, 2016-08-03:
there's actually a linux module for that:

https://github.com/torvalds/linux/blob/master/drivers/tty/serial/sc16is7xx.c

not sure why its not available in raspbian though

falkena, 2017-03-07:
Should be already here:
https://github.com/raspberrypi/linux/issues/1594
https://github.com/neuschaefer/raspi-binary-firmware/commit/b044e9a30a8eee00130406ed85c08da5445a0dd4

savageautomate, 2021-01-15:
**Closed.**  No further activity.

As of Pi4J v2.x, IO expansion is provided via pluggable modules and support for specific IO expansion chipsets is outside of the scope of the Pi4J project.

closed 2021-01-15 by savageautomate