Pi4J/pi4j-v1 #238: Require root for pi3

edwardcapriolo, 2016-05-19:
For the PI3 it seems like pi4j requires root, but python apps do not have this limitation. If do not mind pitching in the fix

savageautomate, 2016-05-25:
@edwardcapriolo 

Pi4J *should* already work without root access.  See this post:
http://www.savagehomeautomation.com/projects/pi4j-now-supports-non-privileged-access-no-more-rootsudo.html

Please let me know if its not working for some reason :-)

PS. certain features such as hardware PWM control still require root access.  Basic GPIO usage no longer requires root access.  But here are a few things need to configure your system and application first.

Thanks, Robert

closed 2016-05-25 by savageautomate

edwardcapriolo, 2016-07-01:
Can you re-open. The problem is this code you mention is only in your unpublished-to maven branch. When are you planning on publishing this release to central?

savageautomate, 2016-07-01:
The SNAPSHOT builds are published to a Maven repository, just not Maven Central:
http://pi4j.com/download.html#Source_Repository

I am working to get the release wrapped up in the next week or two.

edwardcapriolo, 2016-07-01:
I understand about the SNAPSHOT, but the problem is you can not release a project that has a snapshot dep.

savageautomate, 2016-07-01:
I understand :-)   And I agree that its been way too long without a release build. 
Working on final items to release very soon.

Thanks, Robert