Pi4J/pi4j-v1 #241: DS18B20 using Pi4j

Belkheir, 2016-06-03:
Hi,i'm actually working in a project using a raspberry Pi  and temperature sensor DS18B20 , i made some research in internet but i haven't found a way to read temperature data using PI4J instead of  opening the file /sys/bus/w1/devices/28-00*/w1_slave, read it, close it, and then parse the results. i was wondering if there is a way to use the   GpioPinDigitalInput.
Best regards.

savageautomate, 2016-06-03:
Not really.

There is a basic 1-Wire impl in Pi4J.

See this example:
https://github.com/Pi4J/pi4j/blob/develop/pi4j-example/src/main/java/W1TempExample.java

And the impl classes:
https://github.com/Pi4J/pi4j/tree/develop/pi4j-core/src/main/java/com/pi4j/io/w1

But it is based on file I/O.  So it still relies on the one-wire Linux device driver and the file system.

Thanks, Robert

closed 2016-06-03 by savageautomate

Belkheir, 2016-06-03:
Thanks Robert , you're right, I think i found the solution to the issue , i
appreciate your help :)