Pi4J/pi4j-v1 #242: com.pi4j.io.i2c.I2CDevice.write(byte b) does not work on 1.1-SNAPSHOT

tatery, 2016-06-03:
Hi,

Running the following example from 1.1-SNAPSHOT https://github.com/Pi4J/pi4j/blob/develop/pi4j-example/src/main/java/ADS1115GpioExample.java results all the time with the exception: 

`java.io.IOException: Error writing to I2CDevice on I2CBus '1' ('/dev/i2c-1') at address 0x48. Got '-10001'.`

The root of this exception is method: protected int readRegister(int register) 
` device.write((byte)register); <---- root of the exception`

when replacing method `write(byte)` with `write(byte[] buffer, int offset, int size)` then example works correctly.

falkena, 2016-06-12:
Can be resolved by pull request #244. Try current build please.

savageautomate, 2016-06-12:
New 1.1-SNAPSHOT build (which includes this fix) is now published for download.

closed 2016-06-12 by savageautomate