Pi4J/pi4j-v1 #243: Pi4J Examples not working on Pi 3

hankp46, 2016-06-11:
I've tried both the listener and control example and neither have worked on the Pi3. I know my button is working properly since I tested it in another experiment. The only thing I changed is the pin to 26.  Both compiled and run (printout) but no  response from button or blinking led. I'm a novice on Java but think I followed all instructions. Now at standstill. Any suggestions?

savageautomate, 2016-06-12:
I just tested both "Control" and "Listen" examples using Pin #26 on my RPi3B with the latest 1.1-SNAPSHOT and it seems to be working.  

Often the pin numbering is a source of confusion.  When you say pin 26, make sure you are referring to the Pi4J pin numbering scheme pin 26 (_header pin #32_) as shown here: http://pi4j.com/pins/model-2b-rev1.html

hankp46, 2016-06-12:
Thanks, after posting saw your note about pins and got everything working

Hank Poletto
hpoletto@cfl.rr.com

closed 2016-06-12 by savageautomate