Pi4J/pi4j-v1 #246: Setting Pin7 as Input

as-qs, 2016-06-14:
Ever since I upgraded to Jessie and started using kernel 4.4, I've been experiencing some issues working with SPI and GPIO pins.

I seem to have narrowed down this problem to the only input pin I use (GPIO 7 = Pin 7). If I take my code and run it on a wheezy image with the 4.1 kernel, all is fine. However, on the Jessie image with the 4.4 kernel, I am not getting a callback on the listener. 

I think this all has to do with the deprecation of spi-bcm2708, but I'm all out of ideas on what to do.

Any help would be appreciated.

savageautomate, 2016-06-14:
I just tested on a RPi 3B and it seems to be working for me.

Kernel Version:
`Linux rpi3b 4.4.13-v7+ #894 SMP Mon Jun 13 13:13:27 BST 2016 armv7l GNU/Linux`

I have the latest Pi4J 1.1-SNAPSHOT version installed.

I run this sample program:
`sudo pi4j /opt/pi4j/examples/GpioListenExample.java --pin 7 --pull down`

I then test by momentarily connecting and disconnecting a jumper between 3.3VDC and GPIO7.
Here are the results I see:

```
************************************************************
************************************************************
                  <-- The Pi4J Project -->                  
                     GPIO Listen Example                    
************************************************************
************************************************************

------------------------------
|    PRESS CTRL-C TO EXIT    |
------------------------------

 ... Successfully provisioned [GPIO 7] with PULL resistance = [DOWN]

--------------------------------------------------
|  Please complete the [GPIO 7] circuit and see  |
|  the listener feedback here in the console.    |
--------------------------------------------------

 --> GPIO PIN STATE CHANGE: "GPIO 7" <GPIO 7> = HIGH
 --> GPIO PIN STATE CHANGE: "GPIO 7" <GPIO 7> = LOW
 --> GPIO PIN STATE CHANGE: "GPIO 7" <GPIO 7> = HIGH
 --> GPIO PIN STATE CHANGE: "GPIO 7" <GPIO 7> = LOW
 --> GPIO PIN STATE CHANGE: "GPIO 7" <GPIO 7> = HIGH
 --> GPIO PIN STATE CHANGE: "GPIO 7" <GPIO 7> = LOW
 --> GPIO PIN STATE CHANGE: "GPIO 7" <GPIO 7> = HIGH
 --> GPIO PIN STATE CHANGE: "GPIO 7" <GPIO 7> = LOW
 --> GPIO PIN STATE CHANGE: "GPIO 7" <GPIO 7> = HIGH
 --> GPIO PIN STATE CHANGE: "GPIO 7" <GPIO 7> = LOW
 --> GPIO PIN STATE CHANGE: "GPIO 7" <GPIO 7> = HIGH
 --> GPIO PIN STATE CHANGE: "GPIO 7" <GPIO 7> = LOW

```

I also tested with "--pull up" and then connected GPIO7 to a GND pin with the same success results. 

I do have both I2C and SPI enabled via raspi-config.

Can you try this sample code and see if you have any trouble working with GPIO7 independently? 

Thanks, Robert

as-qs, 2016-06-14:
This example works on my end as well; however, once I hook up my custom board to the RPI, it no longer receives anything and stays idle. I tested it with and without my java program running in the background.

Let me provide more details:

I'm using the following pins (pi4j pins):
- 4 relays on pins: 0, 2, 3, 4 (they all work).
- 1 input pin on pin 7, which is driven by the custom board.
- I use pin 10 as my spi cs and set it to low when I need to send something through spi.
- pins 12 13, 14 are used for MOSI, MISO, SCLK.

Thanks for the help.

savageautomate, 2016-06-15:
If you are using the SPI impl in Pi4J, I don't think you have to manage the CS (chip-select) pin (GPIO 10) in your code.  I believe the underlying SPI DEV driver does this for you.

What you are doing seems pretty simple and straightforward ... so I'm not sure what could be going wrong without being able to test/meter/scope the custom board.  

Does the custom board send a 3.3VDC signal to GPIO 7 to trigger it or does it send a GND signal?  Are you using the internal pull up/down resistors on the RPi board or does your board have pull up/down resistors physically wired on the board?

as-qs, 2016-06-15:
The reason I manage the cs line is because that's my way of telling the other board that there is data to be read. It's a more reliable approach that we've come up with in the past year or so as we had a lot of stability issues. I set pin10 low when I'm sending something and bring it back up as soon as I'm finished.

Pin7, which I'm using as input, goes low when there is data available to be read. 

The custom board has pull up/down resistors of its own, and those might be the source of the problem as we've had them changed and modified a few times in the recent months. The crazy thing is that this combination works perfectly with the 4.1 kernel. 

This is how I'm dealing with all the output pins:
`
PinState pinState;
                
if (stateLow == true)
{
    pinState = PinState.LOW;
}
else
{
     pinState = PinState.HIGH;
}

gPin.gpioOutput = gpio.provisionDigitalOutputPin(RaspiPin.getPinByName(pinNumberString), pinName, pinState);`

And this is how I setup the only input pin:
`
gPin.gpioInput = gpio.provisionDigitalInputPin(RaspiPin.getPinByName(pinNumberString), pinName, PinPullResistance.PULL_DOWN);`

I've tried it with pull down, pull up, and none. 

Today I'm planning on measuring the voltage on all the pins with 4.1 and 4.4 kernels. Then I'll hook them up to the custom board and re-do the measurements. Please let me know if you can think of a smarter test.

Thanks again for all your help!

savageautomate, 2016-06-15:
I'm no expert, but your plan sounds like a good place to start.

FYI, on the pin state, you can shorten the conditional block to this just for brevity (no functional difference)
```
PinState pinState = PinState.getState(stateLow);
```

If your board has it's own pull down resistors for the input 7 pin, then you should probably use 
```
PinPullResistance.OFF
```
as not to add extra resistance to the line.  Although like you said that may not really make any real difference in the application.

as-qs, 2016-06-15:
Learned next to nothing doing the "exercise." Here is what was observed:

1) MOSI & MISO get set to 3.2V when the host board is attached; otherwise, they are zero.
2) When the host board is NOT attached, GPIO 7 (pin 4) is set to 3.1V when my program is not running. And it is set to zero when the program is running, which might be because my program sets it as input with pull down. It is set to ~3.1V when the host board is attached with & without my program running.
3) All other pins get identical readings in all every configuration. 

Raw Data:
[RPI_Measurements.xlsx](https://github.com/Pi4J/pi4j/files/317214/RPI_Measurements.xlsx)

as-qs, 2016-06-16:
**Status update:**

I modified my code, so that instead of waiting for a callback on GPIO7, I would continuously send data through SPI (sending 0xFF's). The result is that I am now able to communicate with the SPI and get the system to function semi-correctly. This means the problem is not SPI related and is isolated to the input pin4 (GPIO 7). 

Now, the big question is, what the hell is wrong with that pin??

savageautomate, 2016-06-16:
> 2) When the host board is NOT attached, GPIO 7 (pin 4) is set to 3.1V when my program is not running. And it is set to zero when the program is running, which might be because my program sets it as input with pull down. 

Agreed, this is because your program is provisioning the pin with a PULL-DOWN settings.  The internal GPIO resistors are pulling the voltage down to zero (GND). 

> It is set to ~3.1V when the host board is attached with & without my program running.

So its sounding like a hardware problem to me.  The fact that you are getting a different result when the custom board is attached versus when not attached.  

So first question I think of, is .. is the custom hardware board supposed to be pulling GPIO7 UP (3.3VDC) or DOWN (0VDC GND)?   If it is designed/wired to pull up, then I think you may have a competing issue where the hardware is trying to pull up and the software configured GPIO is trying to pull down.  

If the hardware is designed to pull UP (which is a very common practice), then your program may just need to be modified to handle the inverse logic of the GPIO pin state.  (i.e. signaling your program on a LOW state instead of a HIGH state.)    Of course, I don't know what the rest of the circuit does, so I'm just guessing in the blind :-)

as-qs, 2016-06-16:
Well, the hardware is supposed to set the pin to low when there is data available to be read. And it does that with the 4.1 kernel. So, why is it not doing it with 4.4? What's different?

Ideally, I want to follow the best practices and do this right. However, I also would like to avoid costly hardware revisions if that's possible. It's a dilemma... 

For now, I'm gonna try to focus on figuring out the precise cause of this phenomenon.

savageautomate, 2016-06-16:
> Well, the hardware is supposed to set the pin to low when there is data available to be read

OK, then that sounds like its designed to PULL UP in the hardware when no software is attempting to do anything, which makes senses.  

Who or what piece of code is setting the GPIO7 state to LOW when data is available to be read?  Is this your Java code using Pi4J?

as-qs, 2016-06-16:
No. That's done by the custom board (PIc18). I only read the state of that pin since it's an input pin.

savageautomate, 2016-06-16:
OK, if your code is provisioning the pin with PULL DOWN, then that could be a potential problem.  But I have no idea why it would work on a previous kernel.    And I think you said you already tried changing that.  The pull resistance should probably be set to OFF during the pin provisioning code if the hardware board has its own pull up resistors. 

>  gPin.gpioInput = gpio.provisionDigitalInputPin(RaspiPin.getPinByName(pinNumberString), pinName, PinPullResistance.PULL_DOWN);

savageautomate, 2016-06-16:
Try changing the GPIO7 pull resistance to off and re-measure the voltage with the board attached and with the software running and not running and see if there is any difference.  Previously this measured ~3VDC in both cases.

as-qs, 2016-06-16:
It shows 3.26 and stays that way after running the program. This is with the pull resistance being set to off.

savageautomate, 2016-06-17:
OK, so I'm at a loss as what could be wrong. Is there a way to measure the state of the communications signaling pin on the custom board without it being connected to GPIO7?

as-qs, 2016-06-17:
I'll have to fabricate something to do that test. I'm gonna try to borrow an oscilloscope and see if I can make any sense out of what's going. I do very much appreciate all of your help and the amount of time you have put into this project. 

Even if I don't get this thing fixed, I'm very much interested in figuring out what's really causing all of this headache.

tschakaa, 2016-06-27:
I have the same Problem with a Raspberry Pi 3: The LED lamp indicates that Pin 7 is always on. Any solutions yet - I would like to [control 2 Motors](https://github.com/Pi4J/pi4j/blob/develop/pi4j-example/src/main/java/StepperMotorGpioExample.java) with this code but get stuck due to this problem.

as-qs, 2016-06-27:
No. I contacted the EE that created the custom board to see if he has any ideas, but he hasn't found anything yet. I'll be sure to post here once I have a solution.

Another member (bennuttall) has a similar problem, but I don't think he is using pi4j.

as-qs, 2016-08-23:
I just want to report this problem as resolved. By reducing the SPI speed from 390,000 to less than 100,000, everything is started working as normal. Apparently the new drivers don't like the higher speeds. Such a simple solution for such an annoying problem...

closed 2016-08-23 by as-qs